esp-rs/esp-wifi-sys #1: Can't connect to any AP?

MabezDev, 2022-03-29:
I've been playing around trying to get this work. Scanning works perfectly, I can see a table of networks near me, but If I try to connect to any (I've tried 4 in total, on two different esp32c3's), the WiFi state never transitions to `STA_CONNECTED`. Judging by the logs below is transitions to `STA_STOP`.

```
wifi_connect returned 0
D (405) wifi:scan end: arg=0x0, status=0, ss_state=0x03
V (405) wifi:hint scan, stop scan
V (405) wifi:back home chan=<1,0>, current chan=<10,0>
V (408) wifi:ht20 freq=2412, chan=1
D (412) wifi:filter: set rx policy=4
D (412) wifi:first chan=1
V (416) wifi:scan_done: arg=0x0, status=0, cur_time=647416, scan_id=128, scan state=00
V (423) wifi:call scan_done cb, arg=0x0
D (427) wifi:handoff_cb: status=0
D (431) wifi:clear rc list
D (431) wifi:clear blacklist
D (435) wifi:send disconnect event
D (438) wifi:connect status 1 -> 3
D (442) wifi:disable connect timer
D (443) wifi:clear scan ap list
```

## Things I've tried

- Different AP's
- Different boards
- Changing the country code in `wifi_init`

I've sniffed the connect packed with my laptop in promiscuous mode and the packet seems normal.

bjoernQ, 2022-03-29:
Ah I'm a fool: Please change line https://github.com/bjoernQ/esp32c3-wifi-rs/blob/062cb1aa81a0b0faf8e7f0f4c857dd97b05cbe7a/src/wifi/mod.rs#L425 to channel 0 and try again

bjoernQ, 2022-03-29:
That strange - from your logs it looks like it just tries to scan channel 1 and gives up then - it should scan up to 13 channels I guess (nchan in country config)

MabezDev, 2022-03-29:
It's very strange, it still only scans channel 1 even if I supply `wifi_scan_method_t_WIFI_ALL_CHANNEL_SCAN` in the wifi config for connect :thinking:.

PR #2 (2022-03-29): Leave channel unspecified in configuration
So the scan searches for AP's in all channels until it is found.

Closes #1
closed 2022-03-29 by bjoernQ