esp-rs/esp-wifi-sys #16: ESP32: Wifi logs cannot be configured

bjoernQ, 2022-04-07:
Calling `esp_wifi_internal_set_log_level` or `esp_wifi_internal_set_log_mod` crashes immediately or cause crashes later (depending on when those are called) - so Wifi logs are never enabled on ESP32 (i.e. it works fine on ESP32C3)

Additionally there is a problem with VaList on ESP32 (or Xtensa in general) - that's why it's conditionally compiled only for ESP32C3 in `log_writev`

MabezDev, 2022-04-07:
I've been applying [this patch](https://github.com/esp-rs/rust/commit/ca1979d777822d9e145b8c84e44d1b6b01643214) for vaarg support. I've never personally used vaargs, so it's quite possible this patch is buggy, or is missing some other steps?

bjoernQ, 2022-04-07:
Awesome - will give it a try. Long time since I built the compiler myself the last time 😄

MabezDev, 2022-04-07:
Just to be clear, it's already included in the esp-* branch patch set, so it should already be working.

bjoernQ, 2022-04-11:
The `va_list` implementation seems to be fine since I can use it with some C code written myself and compiled with GCC. However, the `va_list` I get from the WiFi driver looks different. For now, I just print the format string on ESP32 which is better than having nothing. Still an interesting and valid issue so keep it open

bjoernQ, 2023-05-16:
Unfortunately https://github.com/esp-rs/rust/issues/171 doesn't fix this

bjoernQ, 2023-05-17:
Alternatively, we could implement `syslog`, `log_write` and `log_writev` in C until a fix is available.

karlri, 2023-06-29:
This is an ugly hack but it works. It's not very invasive because it replaces only exactly the function call that bugs out. Not likely woth a PR but if anyone is interested in trying it out, here it is (implemented for esp32 only): https://github.com/esp-rs/esp-wifi/commit/8befe40fc4b859d95d877c95af8b4f810ed6ce06

PR #18 (2022-04-11): At least some wifi logs on ESP32
Again, not much to review. Just want to make sure it doesn't break anything for anyone.

This enables WiFi logs for ESP32. It just logs the format string of the log message since there is a problem with the va_list - but that is much better than having nothing logged

This partially resolves #16
PR #293 (2023-10-20): Enable `vsnprintf` on compilers newer than `1.73.0.1`
A not terribly complex patch to resolve #16. It might be easier to just bump MSRV, but 🤷‍♂️.

Also closes https://github.com/esp-rs/esp-mbedtls/issues/13
closed 2023-10-20 by bjoernQ