esp-rs/esp-wifi-sys #26: Isolate the allocator into an esp-alloc crate

ivmarkov, 2022-05-01:
Having a heap allocator in `no_std` is useful with or without `esp-wifi`. Yet, currently, the Rust allocator is implemented as part of `esp-wifi`, on top of a simplistic `malloc` and `free` allocator. Ideally, we should turn it upside down:
* Isolate the allocator in an `esp-alloc` crate. I'm thinking, maybe we can re-use there the [linked-list-allocator crate](https://github.com/phil-opp/linked-list-allocator) which is also used by the [cortex-m community](https://github.com/rust-embedded/alloc-cortex-m/blob/master/Cargo.toml#L27).
* Layer `malloc`, `calloc` and `free` implementations on top of the Rust allocator. These can still stay in `esp-wifi`, as I don't see anything else re-using them so far
* (Optional) Use the `linked-list-allocator` with critical sections, instead of [spin-locks](https://github.com/phil-opp/linked-list-allocator/blob/master/src/lib.rs#L28). This would allow us to use the allocator from ISR contexts as well. 
  * Why would we want this? `embassy-rs` - although being an async framework (i.e. cooperative task scheduling) - has task pre-emption capabilities, by simply allowing the user to run more than one executor. [The main executor runs in the non-ISR context, while 0 or more executors run in ISR contexts and are triggered by software interrupts](https://github.com/embassy-rs/embassy/blob/master/examples/nrf/src/bin/multiprio.rs#L123). This way, tasks in the non-ISR executor can be preempted by tasks scheduled on the e.g. ISR1 executor if a task in it is awoken and scheduled; which - in turn - can be preempted by tasks running e.g. on the ISR2 executor if a task in it is awoken and scheduled. Neat. And all in a single stack! (By the way, [my own little `no_std` compatible adjustment of smol's `async-task` executor should have the same capabilities](https://github.com/esp-rs/embedded-svc/blob/master/src/utils/asyncs/executor.rs#L194) - to be checked in practice.)
  * For maximum performance, we need to protect the allocator with a critical section which is masking **only** the software interrupts that are configured to run the ISR executors from above. This way hardware interrupts (e.g. GPIO button) that run in a no-alloc context can still interrupt and preempt execution of the software ISRs, while these are inside the potentially O(n) alloc-ing code. (Here I have to check how to configure and signal software interrupts on xtensa and riscv for the ESP though - no idea currently!)
  * As per above, we also need an easy way to raise a software interrupt. This way, the "button" GPIO ISR (which has the highest priority, runs in a no-alloc context and is totally unsafe) will only "awake" a regular async task. If this task happens to be scheduled on the ISR1 or e.g. the ISR2 executor, calling its waker will not just schedule the task on the executor queue, but [will also trigger an ISR1 or an ISR2 software interrupt](https://github.com/embassy-rs/embassy/blob/master/embassy-nrf/src/gpiote.rs#L128), and once the button ISR is complete, ISR1 or ISR2 will start executing, with the task that does the button debouncing logic.

bjoernQ, 2022-05-02:
I totally agree - I just forgot to open an issue for it 😄 

I remember that choosing a fitting existing allocator got me some troubles but I don't remember which. Definitely the allocator found here is not really what I want to have.

jessebraham, 2022-05-13:
I have extracted the allocator into a separate crate,  though I have not made any actual changes to the code yet. Repo can be found here: https://github.com/esp-rs/esp-alloc

Additionally opened a PR including said crate: https://github.com/esp-rs/esp-wifi/pull/31

I will likely transfer this issue to the new repository once the aforementioned PR is merged.