esp-rs/esp-wifi-sys #27: Remove the embedded-nal / smoltcp-nal dependencies from `wifi_interface.rs` and esp-wifi

ivmarkov, 2022-05-01:
While `smoltcp` itself has no alternative in the embedded world, I'm less than convinced that `embedded-nal` is the no-brainer for embedded networking: it still relies on the `nb` "async" approach, which is essentially just busylooping with constant polling of the networking layer. While this might have been the norm a few years ago, it is no more:
* `embedded-hal` is moving away from `nb` in favor of the [async-based traits](https://github.com/rust-embedded/embedded-hal/tree/master/embedded-hal-async)
* `smoltcp` itself has an async support and an [`async` feature](https://github.com/smoltcp-rs/smoltcp/pull/394) since quite some time, so it is very easy to bind it to an async executor of your choice, if you prefer so

In my opinion:
* We need to remove from `esp-wifi` all dependencies / reliance on `embedded-nal` / `smoltcp-nal`. This stuff is anyway a thin wrapper on top of the heavy lifting done by `smoltcp` itself
* The [`Wifi` struct](https://github.com/esp-rs/esp-wifi/blob/main/src/wifi_interface.rs#L28) can instead take directly the smoltcp IP interface, i.e. the ["ethernet"](https://github.com/esp-rs/esp-wifi/blob/main/src/wifi/utils.rs#L69) var
* This way, the [`create_network_stack`](https://github.com/esp-rs/esp-wifi/blob/main/src/wifi/utils.rs#L46) utility will construct and return directly the smoltcp interface instead of the `embedded-nal` wrapper. Perhaps, we should rename the utility to `create_network_interface` or suchlike
* The [`network_stack` method of the `Wifi` struct](https://github.com/esp-rs/esp-wifi/blob/main/src/wifi_interface.rs#L36) should instead directly return the native smoltcp interface

With that little surgery, users would be free to either use smoltcp-nal on top of the smoltcp interface returned by `Wifi`, or the smoltcp built-in async facilities, with an async executor of their choice.

bjoernQ, 2022-05-02:
My intention with `embedded-nal` was to just introduce some quality-of-life improvement - so I added it as a non-default feature.
I personally think that `embedded-nal` is the easier API when going the blocking way. But maybe it's just me.

Big thanks for pointing out the async feature of `smoltcp` - I somehow totally missed that. Really need to look into that. That's very exciting.

I don't have a strong opinion on `embedded-nal` but since it's a non-default feature it shouldn't hurt too much.
But I definitely like to have a similar way to configure `smoltcp`. And probably also re-add a `smoltcp` example.

ivmarkov, 2022-05-04:
> My intention with `embedded-nal` was to just introduce some quality-of-life improvement - so I added it as a non-default feature. I personally think that `embedded-nal` is the easier API when going the blocking way. But maybe it's just me.

I completely agree that was a great first step, for a demo etc. The question is - how to continue from here. My problem with `embedded-nal` is that I actually don't see it as a solution, neither for blocking, nor for nonblocking access:
* Issues with blocking access
  * How can I even use a blocking networking API (`embedded-nal` or not), given that on "bare metal" I don't have multithreading, with a preemptive thread scheduler (i.e. FreeRTOS or similar)? This will likely not work even for toy projects 
* Issues with non-blocking access
  * The `poll` based non-blocking access of `embedded-nal` suffers from the fact that it does not gave the notion of a "waker". So the only choice users are having is to poll in a loop, with some sort of arbitrary sleep interval between the polls. I think this is what you are doing in the current examples. It is just that nowadays, with async wakers being built-in in smoltcp, and at least two executors to pick and port (`embassy` and my own little adjustment of smol's `async-task`), I don't see why we would consider the `embedded-nal` model
  * The `nb`-based `embedded-nal` polling model has usability issues as well, in that it does not compose and I cannot write regular secuential code as when using `await`

> I don't have a strong opinion on `embedded-nal` but since it's a non-default feature it shouldn't hurt too much. But I definitely like to have a similar way to configure `smoltcp`. And probably also re-add a `smoltcp` example.

But my point is - your `embedded-nal` configuration is in fact 99% percent `smoltcp` configuration actually. Only where you wrap the smoltcp interface in embedded-nal `NetworkStack` it becomes an embedded-nal configuration. Why don't we keep the existing smoltcp interface configuration, and just return it? Folks can then either use it with async, or layer on top of it embedded-nal if they wish so?

ivmarkov, 2022-05-04:
In other words, just delete [this line](https://github.com/esp-rs/esp-wifi/blob/main/src/wifi/utils.rs#L93), don't create a `WifiClock` and return the `ethernet` var.

MabezDev, 2022-05-04:
I'm only really familiar with smoltcp, embedded-nal didn't exist when I was contributing to smol :D. I think you make some good points @ivmarkov. I suppose the only thing we can look at is what crate support will having embedded-nal implementations get us?

From [the reverse dependencies](https://crates.io/crates/embedded-nal/reverse_dependencies), 17 crates use embedded-nal. Of those 17, 3 are implementations of thr traits so we can take that away. So having embedded-nal trait implementation gives us access to 14 crates of varying quality.

If it's easy to maintain embedded-nal behind a feature, then it probably makes sense to do so, but we should also offer helper methods for using smoltcp directly where appropriate.

bjoernQ, 2022-05-04:
> If it's easy to maintain embedded-nal behind a feature, then it probably makes sense to do so, but we should also offer helper methods for using smoltcp directly where appropriate.

That's basically what I wanted to say 😄

ivmarkov, 2022-05-04:
Well, if we believe that the existing total of 14 lines ([this one](https://github.com/esp-rs/esp-wifi/blob/main/src/wifi/utils.rs#L93) and [this one](https://github.com/esp-rs/esp-wifi/blob/main/src/wifi/utils.rs#L99) are worth a feature, then sure, let's keep it as some sort of utility code.

All I'm asking for is that the `Wifi` interface and struct **not** to depend on `embedded-nal` + `smoltcp-nal` and *just to return a plain smoltcp interface*. Users can always use the above 14 lines utility to wrap the returned smoltcp interface. Or not use them, as I would do.

bjoernQ, 2022-05-04:
So basically, that means that the feature `embedded-svc` shouldn't depend on `embedded-nal`? 
I like the suggestion in general.

We only need to figure out how to deal with `ClientIpStatus::Waiting` and `ClientIpStatus::Done` - technically those relate to the networking stack, not the wifi interface. t.b.h. I don't like that much - but it's the way it is now

So, we need to add some code to handle DHCP (basically what `smoltcp-nal` does for us currently) there using `smoltcp`

Then I'm not sure if we can construct a `smoltcp-nal` instance with an already assigned IP address _if_ we want to keep the feature

MabezDev, 2022-05-04:
Just as an FYI, I also saw they've added [embedded-nal-async](https://github.com/rust-embedded-community/embedded-nal/tree/master/embedded-nal-async) very recently.

ivmarkov, 2022-05-04:
> Just as an FYI, I also saw they've added [embedded-nal-async](https://github.com/rust-embedded-community/embedded-nal/tree/master/embedded-nal-async) very recently.

6 hours ago, to be precise? :D

bjoernQ, 2022-05-04:
> Isn't smoltcp also a networking stack?

It is!

> I think we are getting at a point where some prototyping / experimentation is necessary, English words become difficult. :)

Hope to get to this next week - I think I got your points ;) ... if anyone gets bored meanwhile - feel free to jump in

ivmarkov, 2022-05-04:
I was taking a quick look at the [embedded-nal-async](https://github.com/rust-embedded-community/embedded-nal/tree/master/embedded-nal-async) traits, and they are an exact replica of the older `nb` ones, except done with async futures based on GATs (just like `embedded-hal-async` and everything async-related in `embedded-svc`). 

Nothing wrong with that and quite the opposite - this is a great start! And a way for us to implement e.g. an async mqtt client which works on bare-metal (`esp-wifi` + `esp-hal`) as well as in ESP-IDF (`esp-idf-sys` + `esp-idf-hal` + `esp-idf-svc`), as I can (I think) quite easily map the `embedded-nal-async` traits to ESP-IDF LWIP via Rust STD and `smol`'s async reactor.

But this new development actually reinforces my view that `embedded-nal` and especially `smoltcp-nal`'s `NetworkStack` should probably not be part of the `Wifi` trait impl, because `NetworkStack` is all about the old `nb` based `embedded-nal` and not the new async one.

That is, unless someone comes with a clever PR to `smoltcp-nal`, where the existing `NetworkStack` struct is **extended** to *also* implement the new async traits, by utilizing the `async` feature of smoltcp!

ivmarkov, 2022-05-05:
FYI a summary of a discussion yesterday on the embassy Matrix channel:

@dirbaio pointed out the obvious (in retrospective) - the new `embedded-nal-async` traits have a design flaw where you can't send/recv on multiple sockets simultaneously, because [the api takes the "stack" as a `&mut self`](https://github.com/rust-embedded-community/embedded-nal/blob/master/embedded-nal-async/src/stack/tcp.rs#L56).

There are other design issues as well, as in [accept](https://github.com/rust-embedded-community/embedded-nal/blob/master/embedded-nal-async/src/stack/tcp.rs#L124) returning a fresh socket instance where to await on it *and* on the previously accepted sockets you have to either use [FuturesUnordered](https://docs.rs/futures/0.3.21/futures/stream/futures_unordered/struct.FuturesUnordered.html) (as [I do](https://github.com/ivmarkov/ruwm/blob/master/ruwm/src/web.rs#L79)), or have a trait abstraction for a "spawner" ([like this one](https://github.com/esp-rs/embedded-svc/blob/master/src/executor.rs#L8)) so that you can spawn a new task which would process the newly-accepted socket.

Both of these (`FuturesUnordered` and traitifying the notion of a Spawner) are incompatible with embassy-rs's executor as they do require allocations, while embassy is a completely static no-alloc thing. There's also an issue with the wakers of `FuturesUnordered` being a special thing incompatible with the embassy ones, but that's a whole other can of worms.

Perhaps an easier approach which is also future-compatible with `embassy` would be the one discussed a few days ago on the Matrix channel - standardize at least the notion of asynchronous `no_std` Reader/Writer, then write e.g. async MQTT client, HTTP client / server *only* with these traits, where it is up to the user of e.g. the HTTP server to spin the socket accept loop in whatever platform-specific way that happens. Effort to have a common set of `no_std` async Reader/Writer (and sync ones too) [here](https://github.com/embassy-rs/embedded-io/blob/master/src/async_.rs).

So FWIW and IMO we should continue for now with the original plan of removing the dependency on `embedded-nal` from the `Wifi` trait and instead having it depend directly on the smoltcp interface notion. We can still have utilities for e.g. asynchronous work with smoltcp, as in opening a socket, but these likely will be executor-specific ([here's the embassy one](https://github.com/embassy-rs/embassy/blob/master/embassy-net/src/stack.rs#L65)), as I don't yet see a way to come up with a set of networking traits than can abstract over the no-alloc Embassy executor on one hand, and the alloc ones like [my own creations](https://github.com/esp-rs/embedded-svc/blob/master/src/utils/asyncs/executor.rs) + the standard ones that do work on the ESP-IDF (futures-rs, smol, etcetera) on the other.

ivmarkov, 2022-05-05:
@lulf ^^^ I'm very interested in your comments / suggestions/ corrections as well.. Shall we move this discussion to `embedded-hal`?

lulf, 2022-05-05:
@ivmarkov I think your comment sums it up well! I think there is probably some changes we can make to the embedded-nal-async traits to have the 'shared stack' situation work, but I've not played enough with that concept yet and focused on 'client usage' for now. No matter what is done with embedded-nal-async, I think it's good to keep any http, mqtt, tls client work with just the bare Read/Write async traits from embassy (soon moved to embedded_io).  Then, embedded-nal-async support might be a nice convenience for some, but not 'forced' on anyone if made 'optional'.

ivmarkov, 2022-05-05:
I think we have to solve the "shared stack" issue for clients as well, or else how can I have an mqtt client and an http client simultaneously, each spinning its own async loop and its own tcp socket?

PR #29 (2022-05-10): Favor smoltcp
This fixes #27 and #28 

This PR changes the user facing networking API back to `smoltcp`.

Keeping `embedded-nal` as an optional feature turned out to be a bit tricky so support is dropped for now.

Additionally this introduces
- esp-println
- esp-backtrace

as dev-dependencies and removes the `println` frunctionality from the crate (since it's not the right place for something like that)
closed 2022-05-10 by bjoernQ