esp-rs/esp-wifi-sys #37: Feature Request: UDP Sockets

D1plo1d, 2022-06-29:
I'm interested in using a esp32c3 to received UDP packets. I've been playing with the DHCP example and it looks like it is working quite well (kudos!) and I'm excited to use this more - is there any chance that UDP support is coming soon?

MabezDev, 2022-06-29:
UDP should be fully supported. We use [smoltcp](https://github.com/smoltcp-rs/smoltcp) as our network stack, `esp-wifi` implemented smoltcp's `Device` trait allowing smoltcp to leverage the onboard WiFi :).

closed 2022-06-29 by MabezDev

MabezDev, 2022-06-29:
Can you post the full code, sounds like a setup issue. Consider joining the smoltcp matrix: https://matrix.to/#/#smoltcp:matrix.org, very friendly folks in there :).

D1plo1d, 2022-06-29:
@MabezDev will do! I've managed to get it a bit closer to working - I can open the UDP socket now (although it's involves creating and throwing away a TCP socket first because otherwise the socket set complains about being full) but I can't seem to receive any UDP packets yet. Here's my full code:

https://github.com/D1plo1d/raves-on-rust/blob/main/esp32c3_osc_receiver/src/main.rs#L39

D1plo1d, 2022-06-29:
I found my issue - little logic mistake on my part where I stopped listening for UDP packets almost immediately after opening the socket :sweat_smile: - anyways it works now! Thanks for pointing me in the right direction! ^_^