esp-rs/esp-wifi-sys #47: Provide source code of all libraries instead of providing a static linked library

luojia65, 2022-08-19:
By now if we need to look into what this library includes, we have to do reverse engineering works. By providing source code (or write a clean room one without reading its original source) we can get this library bottom-up in Rust compiler only without tampering with the toolchains.

MabezDev, 2022-08-19:
Unfortunately for the time being this is fairly unlikely to happen. To be honest, I'm not aware of _any_ open source WiFi stack, this is just the way it is currently. You can follow https://github.com/espressif/esp32-wifi-lib/issues/2 for more updates. In general we try and move everything we possibly can out of the blobs that don't _have_ to be there.

MabezDev, 2024-02-20:
Closing, please follow the upstream issue.

closed 2024-02-20 by MabezDev