esp-rs/esp-wifi-sys #57: Code fails to compile on latest `main`

TheButlah, 2022-09-15:
Version: `69956d42d2ddf00dd7021e429ed815f982e35319` (latest main)

Error:
```
error[E0433]: failed to resolve: could not find `wifi_interface` in the crate root
   --> /Users/ryan.butler/.cargo/git/checkouts/esp-wifi-836f3b2af57fa847/69956d4/src/wifi/utils.rs:103:31
    |
103 |     interface: RefCell<crate::wifi_interface::Wifi<'a>>,
    |                               ^^^^^^^^^^^^^^ could not find `wifi_interface` in the crate root

error[E0433]: failed to resolve: could not find `wifi_interface` in the crate root
   --> /Users/ryan.butler/.cargo/git/checkouts/esp-wifi-836f3b2af57fa847/69956d4/src/wifi/utils.rs:110:27
    |
110 |         interface: crate::wifi_interface::Wifi<'a>,
    |                           ^^^^^^^^^^^^^^ could not find `wifi_interface` in the crate root

error[E0433]: failed to resolve: could not find `wifi_interface` in the crate root
   --> /Users/ryan.butler/.cargo/git/checkouts/esp-wifi-836f3b2af57fa847/69956d4/src/wifi/utils.rs:122:31
    |
122 |         F: FnOnce(&mut crate::wifi_interface::Wifi<'a>) -> R,
    |                               ^^^^^^^^^^^^^^ could not find `wifi_interface` in the crate root

error[E0392]: parameter `'a` is never used
   --> /Users/ryan.butler/.cargo/git/checkouts/esp-wifi-836f3b2af57fa847/69956d4/src/wifi/utils.rs:102:20
    |
102 | pub struct Network<'a> {
    |                    ^^ unused parameter
    |
    = help: consider removing `'a`, referring to it in a field, or using a marker such as `PhantomData`

error[E0392]: parameter `'n` is never used
   --> /Users/ryan.butler/.cargo/git/checkouts/esp-wifi-836f3b2af57fa847/69956d4/src/wifi/utils.rs:166:23
    |
166 | pub struct Socket<'s, 'n: 's> {
    |                       ^^ unused parameter
    |
    = help: consider removing `'n`, referring to it in a field, or using a marker such as `PhantomData`

Some errors have detailed explanations: E0392, E0433.
For more information about an error, try `rustc --explain E0392`.
error: could not compile `esp-wifi` due to 5 previous errors
```

bjoernQ, 2022-09-15:
I guess it's indeed the HAL version - would you mind to try the https://github.com/esp-rs/esp-wifi/tree/esp32s3-wifi branch - in that branch it's already on 0.2 for esp32c3-hal

TheButlah, 2022-09-20:
Doesn't appear to fix it. I also tried the latest main (which is 5 commits ahead of that branch) and that fails too.

It looks like the `utils` feature ought to also toggle on the `embedded-svc` feature, but it doesn't currently.
Should be a simple fix, I'll open a PR

jessebraham, 2022-09-20:
I just tested and merged a PR this morning and it was working fine, so not sure what's going on. I also just did a clean clone of the repository on a different system than I usually develop on and it's working fine.

TheButlah, 2022-09-20:
> I just tested and merged a PR this morning and it was working fine

What feature flags do you pass? If I pass `embedded-svc` it works fine, but that isn't on in the default features.
However, the `utils` feature is on by default, and the code in `wifi/utils.rs` actually references `crate::wifi_interface`, which is feature gated on `embedded-svc`

PR #64 (2022-09-22): Make utils feature independent of embedded-svc feature
This closes #57 and makes #63 obsolete.

@TheButlah this would make your PR obsolete - mind giving this a try?

This moves some utility code which depends on the `embedded-svc` away from the independent `utils.rs` code
closed 2022-09-22 by bjoernQ