esp-rs/esp-wifi-sys #67: Provide simple web server example

MabaKalox, 2022-10-03:
Consider adding simple web server to examples.

jessebraham, 2022-10-03:
I'm not aware of any `no_std` web servers available on [crates.io](https://crates.io) so we will likely need to implement some web server ourselves, unless there is a package available which I've missed. @bjoernQ can correct me if I'm wrong about this.

I do agree though, eventually it would be nice to demonstrate this functionality.

MabaKalox, 2022-10-03:
I mostly ask it, because I feel totally lost. And have no idea where to start. But for sure will send pull request... if manage to write something.
BTW: smoltcp has an [example](https://github.com/smoltcp-rs/smoltcp/blob/master/examples/server.rs) which called server, but for now I am too dump to understand, what it actually does.

bjoernQ, 2022-10-04:
I also don't know any no-std/no-alloc HTTP server but probably writing something very simple should be possible in a few lines of code. (While a real HTTP server is definitely a lot of code)

In general I think at least some example showing a TCP server would be good and something that "looks like" HTTP is probably convenient since it can be tested in a brower.

taunusflieger, 2022-10-12:
I'm also looking for a HTTP server on an ESP32C3. With IDF / RTOS it is available. I'm wondering if it makes sense on no-std without having an async HAL and Wifi - and Embassy as an async application model. Interested to get other people's perspective.

bjoernQ, 2022-12-01:
#91 adds an oversimplified example showing a single server-socket used to do something that "looks like" HTTP - a real HTTP server is definitely out-of-scope for this crate

bjoernQ, 2022-12-01:
Closing since we now have an example using server sockets - as said a real HTTP server is out-of-scope for this crate

closed 2022-12-01 by bjoernQ

MabaKalox, 2022-12-04:
Big thanks!