esp-rs/esp-wifi-sys #70: Stop bundling a global allocator

TheButlah, 2022-10-07:
Looks like `esp-wifi` includes a global allocator. It would be better to let the user choose their own global allocator.

I'll submit a PR to do this.

fenduru, 2022-10-09:
This also prevents me from using esp-wifi in a project freshly created with https://github.com/esp-rs/esp-idf-template as both the std and no-std options there bring in an allocator

MabezDev, 2022-10-10:
> This also prevents me from using esp-wifi in a project freshly created with https://github.com/esp-rs/esp-idf-template as both the std and no-std options there bring in an allocator

@fenduru if you are using an esp-idf based project, you do not need esp-wifi. The esp-idf-* packages already cover WiFi usage.

fenduru, 2022-10-10:
@MabezDev they don't cover BLE

closed 2022-10-19 by TheButlah