esp-rs/esp-wifi-sys #74: Support `direct-boot` mode

TheButlah, 2022-10-19:
So far I've been really appreciating tools like `defmt-rtt`, `cargo embed`, `probe-rs` etc. I believe these tools are only possible with the direct-boot mode of `esp-hal`. Therefore it would be awesome to add support for using `esp-wifi` in `direct-boot` mode.

Does anyone have any guidance for someone looking to implement this? I assume its all to do with how the linker script works.

bjoernQ, 2022-10-20:
I agree it would be nice to make this compatible with `direct-boot` and yes it should be just a matter of adjusting the linker script.

The esp-boot linker script https://github.com/esp-rs/esp-hal/blob/main/esp32c3-hal/ld/bl-riscv-link.x needed a few adjustments to make it work so in theory it's just a matter of configuring the sections similar so that code and data from the blobs go into the right sections.

TheButlah, 2022-10-20:
> needed a few adjustments to make it work
Could you link the relevant PR that added these adjustments or otherwise elaborate a bit for the benefit of anyone looking to implement this?

bjoernQ, 2022-10-20:
This commit https://github.com/esp-rs/esp-hal/commit/033824391dddb2b7e9e201d14d2e587b84dd628d is basically what enabled esp-wifi

bjoernQ, 2022-10-31:
So I tried this a bit and got it to get past the linker step but I wasn't able to get it to work - might look into that again but that experiment wasn't really encouraging

ImUrX, 2022-11-19:
I don't know much about linker scripts, but do you have a guide I could follow or something so I can start trying?
I found a guide on [how linker scripts work](https://users.informatik.haw-hamburg.de/~krabat/FH-Labor/gnupro/5_GNUPro_Utilities/c_Using_LD/ldLinker_scripts.html#Concepts), which is helpful tbh

lucasvr, 2023-10-24:
Hello folks. I have WiFi working in direct-boot mode. I'm not using `esp-rs` nor `esp-idf`, but hopefully this is useful information for you:

1. ESP32-C3 ROM allocates `.bss` and `.data` between 0x3fcd_e710 and 0x3fce_0000 -- and the WiFi driver needs some of that data. This means that the `MEMORY` block in the linker script must assign no more than 377K to both IRAM (at 0x4038_0000) and DRAM (at 0x3FC8_0000), otherwise the OS will corrupt that data.
2. In addition to allocating the relevant sections in IRAM, the data must also be copied to DRAM. This can be done from the `_start` entry point in the asm file. ESP-IDF implements that [right here](https://github.com/esp-rs/esp-hal/blob/29f3518c047fa31ad29a70f0ee73e3484af6c8d8/esp-riscv-rt/src/lib.rs#L336).

jessebraham, 2023-12-04:
Give that we have removed `direct-boot` support in `esp-hal`, I'm not sure it makes sense to pursue this any further. IMO this issue can be closed.