esp-rs/esp-wifi-sys #79: UUID confusion?

brandonros, 2022-10-26:
```rust
ble.cmd_set_le_advertising_data(create_advertising_data(&[
            AdStructure::Flags(LE_GENERAL_DISCOVERABLE | BR_EDR_NOT_SUPPORTED),
            AdStructure::ServiceUuids16(&[Uuid::Uuid16(0x1809)]),
            AdStructure::CompleteLocalName("BLE_TO_ISOTP"),
        ])).unwrap();
```

could you help explain `AdStructure::ServiceUuids16(&[Uuid::Uuid16(0x1809)]),`

versus

```rust
gatt!([service {
            uuid: "0000abf0-0000-1000-8000-00805f9b34fb", // SPP_SERVICE
            characteristics: [
                characteristic {
                    uuid: "0000abf2-0000-1000-8000-00805f9b34fb", // DATA_NOTIFY
                    read: rf,
                },
                characteristic {
                    uuid: "0000abf3-0000-1000-8000-00805f9b34fb", // COMMAND_WRITE
                    write: wf,
                },
            ],
        },]);
```

is the service UUID duplicated?

bjoernQ, 2022-10-27:
The advertising data is just what the device announces and the `gatt!` macro is what the central device will get during discovery.

e.g. 0x1809 is `HEALTH_THERMOMETER_SERVICE` but the example doesn't really have that service. It's unusual but it's just a simple example here

brandonros, 2022-10-28:
does the 0x1809 HEALTH_THERMOMETER_SERVICE need to match the service UUID used in the `gatt!` macro?

bjoernQ, 2022-10-31:
Technically it doesn't need to match. The advertising data is what a device "sees" when scanning for BLE devices while the services it will see during discovery (i.e. when it connects to the device) might be different ones.

Also, the amount of data you can fit into the advertising packet is limited. If you control both ends of the communication you can use whatever works for you for the advertising data. If you don't control the BLE central device then you need to know what the central device is looking for to match that

closed 2022-11-18 by bjoernQ