esp-rs/esp-wifi-sys #87: USB CDC stops working during coex example

hrrs01, 2022-11-23:
I tried to run the coex example on a Beetle esp32c3, when my usb suddenly disconnected, and the next time i tried to connect the esp32 to the desktop, it started giving me FailedEnumeration errors. I suspect strongly its related to something similar to this issue: https://github.com/espressif/arduino-esp32/issues/6264. I reset the chip, by pulling down pin9, and uploading a blinky script, which solved the problem.

For the occational googler, to fix the chip after "bricking" the usb, try pulling gpio9 to ground, while connecting the usb, and upload a different script.

bjoernQ, 2022-11-23:
Can you please try the branch in #86

hrrs01, 2022-11-23:
Its working now

closed 2022-11-23 by hrrs01