esp-rs/esp-wifi-sys #92: Coex support for all chips

jessebraham, 2022-11-30:
- [ ] ESP32
- [x] ESP32-C3
- [ ] ESP32-S2
- [ ] ESP32-S3

elpiel, 2023-04-27:
@bjoernQ not sure what `coex` is but I noticed you removed a few examples in #166 
Is this issues still valid in that case?

bjoernQ, 2023-04-28:
> @bjoernQ not sure what `coex` is but I noticed you removed a few examples in #166 Is this issues still valid in that case?

Thanks for pointing this out.  `coex` means co-existence of Bluetooth and WiFi (using it at the same time)

Removing the `coex` example from ESP32-C6 examples was a mistake but since it's not yet supported, we can re-add it once we add support for it.

UntakerAnonPLAY, 2023-07-28:
So current situation means, that it is impossible to use Bluetooth and WiFi at the same time on ESP32 chip? Or is there a way to implement this feature using rust/mixture of rust and c?

bjoernQ, 2023-08-07:
> So current situation means, that it is impossible to use Bluetooth and WiFi at the same time on ESP32 chip? Or is there a way to implement this feature using rust/mixture of rust and c?

In order to get COEX working we need correct implementations for functions like https://github.com/esp-rs/esp-wifi/blob/ca88c2ef5394bfaf982cec589ff03f90944f9cf3/esp-wifi/src/wifi/mod.rs#L439-L455 and others

IIRC we have some implementations there but just didn't got COEX to work on ESP32

0xkelvin, 2023-10-18:
hello @bjoernQ 

we tried COEX on esp32c6, basically we ported coex from c3, we got the panic as below : 
[image]

it seems the problem is comes from libcoexist.a

we are very keen to use esp32c6 for our telematic project, it is great if we can support and guidance to make it work

thanks

bjoernQ, 2023-10-18:
Seems like it doesn't like the config in src\ble\os_adapter_esp32c6.rs - you can double check it comparing to ESP-IDF's components\bt\include\esp32c6\include\esp_bt.h

I think I already spotted a difference in rtc_freq which should be 16000 but currently it's configured to 32000 (probably copy-paste error). I could imagine the driver won't use it without COEX

Unfortunately, the error code is not documented but even if it was it would probably be a generic ILLEGAL_ARGUMENT kind of error.

tunguyen301096, 2023-10-18:
Hi @bjoernQ ,

It seems like the above error doesn't occur after I double check with ESP-IDF's components\bt\include\esp32c6\include\esp_bt.h and set the config for bt in src\ble\os_adapter_esp32c6.rs

```
pub(crate) static BLE_CONFIG: esp_bt_controller_config_t = esp_bt_controller_config_t {
    config_version: 0x20230113,
    ble_ll_resolv_list_size: 4,
    ble_hci_evt_hi_buf_count: 30,
    ble_hci_evt_lo_buf_count: 8,
    ble_ll_sync_list_cnt: 5,
    ble_ll_sync_cnt: 1,
    ble_ll_rsp_dup_list_count: 20,
    ble_ll_adv_dup_list_count: 20,
    ble_ll_tx_pwr_dbm: 9,
    rtc_freq: 32000,
    ble_ll_sca: 60,
    ble_ll_scan_phy_number: 1,
    ble_ll_conn_def_auth_pyld_tmo: 3000,
    ble_ll_jitter_usecs: 16,
    ble_ll_sched_max_adv_pdu_usecs: 376,
    ble_ll_sched_direct_adv_max_usecs: 502,
    ble_ll_sched_adv_max_usecs: 852,
    ble_scan_rsp_data_max_len: 31,
    ble_ll_cfg_num_hci_cmd_pkts: 1,
    ble_ll_ctrl_proc_timeout_ms: 40000,
    nimble_max_connections: 2,
    ble_whitelist_size: 12,
    ble_acl_buf_size: 255,
    ble_acl_buf_count: 24,
    ble_hci_evt_buf_size: 70,
    ble_multi_adv_instances: 1,
    ble_ext_adv_max_size: 31,
    controller_task_stack_size: 4096,
    controller_task_prio: 253,
    controller_run_cpu: 0,
    enable_qa_test: 0,
    enable_bqb_test: 0,
    enable_uart_hci: 0,
    ble_hci_uart_port: 0,
    ble_hci_uart_baud: 0,
    ble_hci_uart_data_bits: 0,
    ble_hci_uart_stop_bits: 0,
    ble_hci_uart_flow_ctrl: 0,
    ble_hci_uart_uart_parity: 0,
    enable_tx_cca: 0,
    cca_rssi_thresh: (256 - 50) as u8,
    sleep_en: 0,
    coex_phy_coded_tx_rx_time_limit: 0,
    dis_scan_backoff: 0,
    ble_scan_classify_filter_enable: 1,
    cca_drop_mode: 0,  //???
    cca_low_tx_pwr: 0, //???
    main_xtal_freq: 40,
    ignore_wl_for_direct_adv: 0,
    config_magic: 0x5A5AA5A5,

    cpu_freq_mhz: 160,
    enable_pcl: 0, // CONFIG_BT_LE_POWER_CONTROL_ENABLED
    version_num: 0,
};
```

However wifi can't scan and return an error like below picture. Bluetooth is not started advertising and still wait for wifi works properly. Could you help me check with the my logs ?

[image]

tunguyen301096, 2023-10-19:
Hi @bjoernQ ,

After comparing with ESP-IDF and add `G_COEX_FUNCS`, I updated the configuration. Wifi already works, but I got an issue with Bluetooth. The function `ble_hci_trans_hs_cmd_tx` returned `1082268952`. I found this function in `libbt.a`, and it seems like the Rust code didn't call this function correctly.

I guess the code is calling un-correct function, not in `libbt.a`

[image]
[image]

tunguyen301096, 2023-10-20:
After a lot of workaround, I also face the issue related to `r_ble_hci_ram_hs_cmd_tx`, This issue also noted in npl.rs
It seems Wifi can work in Coex, but Bluetooth is having issue.

[image]

[image]

bjoernQ, 2023-10-20:
Does increasing the delay change anything?

tunguyen301096, 2023-10-20:
I tried to increase to `100_000 `, `1_000_000`, `10_000_000`. But still can't

bjoernQ, 2023-10-23:
Is that branch available somewhere? Maybe I can have a peek at it

tunguyen301096, 2023-10-23:
Hi @bjoernQ ,

I have created a draft PR at https://github.com/esp-rs/esp-wifi/pull/300

Could you help me review it ?