esp-rs/esp-wifi-sys #93: Bluetooth support for all chips

jessebraham, 2022-11-30:
- [x] ESP32
- [x] ESP32-C3
- [ ] ESP32-S2
- [ ] ESP32-S3

bjoernQ, 2022-12-01:
I think there is no bluetooth for ESP32-S2?

jessebraham, 2022-12-01:
I just copy-pasted the list for all these issues haha my bad

SergioGasquez, 2023-02-03:
I think we can check ESP32-C2, don't we @bjoernQ?

Finomnis, 2023-09-10:
Do I understand correctly that "Bluetooth support" here is based on `bleps`? Are there plans to release `bleps`, or at least bring it out of "toy state" any time in the near future?

MabezDev, 2023-09-10:
We can probably cut a release of bleps soon, it probably needs a bit of polish with some docs etc. Getting it out of "toy" state requires battle testing the thing, and contributions from the community to fill the gaps.

Finomnis, 2023-09-11:
Then maybe replace the words "toy state" with "early alpha", so people are motivated to contribute. "toy state" sounds to me like it isn't intended to ever be a real thing

bjoernQ, 2023-09-12:
It's important that esp-wifi doesn't depend on the bleps crate. What esp-wifi offers for Bluetooth is an HCI connection.

For very simple use-cases like advertising only you could just pass it the HCI commands needed without using any stack.

Unfortunately, there are not many options besides bleps for a no-std BLE stack currently. One promising crate is https://crates.io/crates/burble - they are considering making it no_std at some point.

Another option would be to port/use Nimble (which is written in C).

Initially bleps was really just meant to be used for the examples - that's why it says it's a toy. I think the biggest thing missing currently is pairing/bonding

JurajSadel, 2023-11-01:
Update: I've started working on adding BLE for H2.

closed 2023-12-06 by bjoernQ