esp-rs/esp-wifi-sys #94: WiFi support for all chips

jessebraham, 2022-11-30:
- [x] ESP32
- [ ] ESP32-C2
- [x] ESP32-C3
- [ ] ESP32-C6
- [ ] ESP32-H2
- [x] ESP32-S2
- [x] ESP32-S3

stevefan1999-personal, 2022-12-07:
I'm not really into these kind of these mixed hardware-software engineering (and failed once to do so) but...any hint on implementing WiFi for ESP32-C2?

We can steal some headers from:
https://github.com/espressif/esp-idf/blob/master/components/bt/include/esp32c2/include/esp_bt.h

And the wireless communication blobs from:
https://github.com/espressif/esp32-wifi-lib/tree/master/esp32c2

And physical layout data from:
https://github.com/espressif/esp-idf/tree/master/components/esp_phy/esp32c2/include

The CSR can also be stolen here:
https://github.com/espressif/esp-idf/tree/master/components/esp_rom/esp32c2/ld

bjoernQ, 2022-12-07:
As you already raised the issue for that, we first need https://github.com/esp-rs/esp-hal/issues/287 in esp-hal to make esp-wifi work for ESP32-C2

The headers and the blobs should be copied / build via https://github.com/esp-rs/esp-wireless-drivers-3rdparty - the Makefile needs to be changed to support ESP32-C2 - not sure if the in mentioned revision of esp-idf there was already C2 support

Once the blobs and headers are copied over the `mkbinding.bat` needs to get adapted (Sorry there is currently just a Windows batch file - mid-term we should replace that with an x-task like binary crate, make sure to use bindgen 0.59.2 for now)

As you already found out the phy-init-data needs to be taken from https://github.com/espressif/esp-idf/tree/master/components/esp_phy/esp32c2/include

Also, as you already found out the needed ROM functions can be found here: https://github.com/espressif/esp-idf/tree/master/components/esp_rom/esp32c2/ld

stevefan1999-personal, 2022-12-07:
@bjoernQ seems like this will be a tad bit more complicated...On ESP32-C2 Espressif is using Nimble on it...that's why it is not a trivial task to port even if it is up for half a year already...

bjoernQ, 2022-12-07:
Nimble is just one of the BLE stacks supported by esp-idf - here we just expose HCI for BLE and let the user use their own stack. (Actually, I don't know of some real world BLE stacks in Rust so the example uses a toy-level implementation of a BLE stack).

Should be possible to add WiFi support even without considering BLE (like it's currently the case for ESP32-S3) - should be also much easier to just consider WiFi for the start since BLE support is much more complex to implement currently

Once you got the headers and bindgen things sorted out, it _should_ just be a matter of
- providing the ROM functions linker script
- adding `common_adapter_esp32c2.rs` and `phy_init_data_esp32c3.rs`
- adding `timer_esp32c2.rs`
- adapt `g_wifi_osi_funcs: wifi_osi_funcs_t` in `./src/wifi/mod.rs`

I'd expect it to be very similar to the ESP32-C3 which should help

bjoernQ, 2023-01-02:
@stevefan1999-personal Are you still interested in working on WiFi for ESP32-C2? If not or if you got stuck: I am considering updating the drivers in the next few days and I can take this over

stevefan1999-personal, 2023-01-02:
@bjoernQ hmm I'm somewhat busy with my life recently to get some company projects done first and so you can take over, I may come look over occasionally though

bugadani, 2023-11-28:
All the check boxes are ticked in the opening message. I there some reason this issue is still open?

closed 2023-11-28 by bjoernQ