esp-rs/esp-wifi-sys #97: Support async

ImUrX, 2022-12-02:
Would be very nice.
I haven't actually read a lot on how it should work. We would first need ``embedded-svc`` to implement an ``async trait`` which is nightly. But I see a lot of calls like scanning that could be async and it would be pretty nice to have that, I also see that ``smoltcp`` implements ``Waker``s for their socket stuff, could we also have async sends and receives :astonished:?

bjoernQ, 2022-12-02:
It's definitely something we will do. 

Currently not exactly a top priority compared with other tasks but it will come

ImUrX, 2022-12-02:
Yeah, obviously supporting the other devices should take priority. Wanted some confirmation from y'all if it was on the list!

MabezDev, 2022-12-18:
This is something I am actively looking into and will be working on in the future.

MabezDev, 2023-02-09:
FYI we've landed two initial forms of async networking, esp-now with #121 and async wifi via embassy_net in #125.

TheButlah, 2023-04-05:
any chance async BLE is next on the horizon? I finished porting my firmware to embassy-net a few days ago and am looking at BLE next ❤️

bjoernQ, 2023-04-05:
> any chance async BLE is next on the horizon? I finished porting my firmware to embassy-net a few days ago and am looking at BLE next ❤️

It's definitely near the top of my personal todo list - I think getting async-io for the HCI stream shouldn't be hard but at least the demo BLE stack used in the demos also need to become async (which I will also look into)

TheButlah, 2023-04-05:
related: https://github.com/bjoernQ/bleps/issues/5

bjoernQ, 2023-04-18:
Async BLE was added in #161

MabezDev, 2023-08-02:
With Async WiFi, BLE & ESP-NOW, we can close this :).

closed 2023-08-02 by MabezDev