esp-rs/esp-wifi-sys #99: Getting linking issues when using esp-wifi

ImUrX, 2022-12-08:
I was adding esp32 support to my esp32c3 project and for some reason I'm getting linking issues when activating the esp-wifi feature I have.
I honestly dont understand much so maybe I can get some help in here:
[This is the linking error](https://gist.github.com/ImUrX/3049de6e08f1ba9dab84c059d68201e5) and [this is the project](https://github.com/ImUrX/SlimeVR-Rust/tree/esp32-support/firmware)

bjoernQ, 2022-12-09:
The linker error doesn't seem to be caused by `esp-wifi` but it's really a weird one

Removing `lto = "thin"` from `firmware/Cargo.toml` makes the build pass for me. Also `lto = "fat"` works for me.

ImUrX, 2022-12-09:
Yeah quite interesting, thanks for noticing this

ImUrX, 2022-12-09:
so when using fat or true, I get:
```
warning: Linking globals named 'DefaultPreInit': symbol multiply defined!

error: failed to load bitcode of module "xtensa_lx_rt-a4b0ba84cc8b089d.xtensa_lx_rt.1ab210e4-cgu.2.rcgu.o": 
```
or ``Linking globals named 'save_context': symbol multiply defined!``
It happens with or without esp-wifi so now I know it's another thing, now I need to find which is the linking script that is being bad

ImUrX, 2022-12-09:
esp32c3 also has the same problem
```
warning: Linking globals named 'error: riscv-rt appears more than once in the dependency graph': symbol multiply defined!

error: failed to load bitcode of module "riscv_rt-b59bab748cac7f4e.riscv_rt.19238726-cgu.10.rcgu.o":
```

stevefan1999-personal, 2022-12-10:
can confirm but I'm building my own firmware:

[image]

bjoernQ, 2022-12-12:
It seems like the linked project really pulls xtensa-lx-rt 0.13.0 and 0.14.0 which explains the duplicate symbols - it's really important to not pull multiple versions of the `rt` crates (for both Xtensa and RISCV)

bjoernQ, 2022-12-13:
I guess we can close this?

closed 2022-12-13 by ImUrX

ImUrX, 2022-12-15:
@bjoernQ i suggest using ``xtensa-lx-rt`` on ``0.13.0`` until ``esp32-hal`` and others update to ``0.14.0``

bjoernQ, 2022-12-15:
Everything should be on xtensa-lx-rt 0.14.0 already in their latest version - if we forgot to update anything please let me know

ImUrX, 2022-12-15:
yeah nvm, i didnt notice there was a new hal version from 3 days ago. I'm still getting a linking error but now it's related to the std ``core`` stuff
```
note: /home/uri/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8_4_0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/uri/proyects/SlimeVR-Rust/firmware/target/xtensa-esp32-none-elf/debug/deps/libcompiler_builtins-7cf6ae26e2cc27f6.rlib(compiler_builtins-7cf6ae26e2cc27f6.compiler_builtins.52d2b8f1-cgu.8.rcgu.o):(.literal._ZN17compiler_builtins3mem7memmove17h94022fc577f39ba1E+0x0): undefined reference to `core::ptr::mut_ptr::<impl *mut T>::sub'
```
And I [no longer see different versions of ``xtensa-lx-rt``](https://gist.github.com/ImUrX/f0dd952bd5e91df9a29322946655f75b) :p