esp-rs/esp-wifi-sys #109: WPA3 support

averyanalex, 2023-01-13:
What about WPA3Personal support? It's [already supported](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/wifi-security.html) in esp-idf.

bjoernQ, 2023-01-13:
To be honest I never tried - I think my equipment doesn't even support it.

In https://github.com/esp-rs/esp-wireless-drivers-3rdparty/blob/f4caebff200e8f6f51b0a11d2b69ca56c76bb1c9/patch/esp32c3/sdkconfig.defaults#LL5-L5C38 we disable `CONFIG_ESP32_WIFI_ENABLE_WPA3_OWE_STA` but we leave `ESP32_WIFI_ENABLE_WPA3_SAE` as default (``y`) - so I'm really not sure

If you or anyone else can test it on your side let us know about the results

averyanalex, 2023-01-13:
> If you or anyone else can test it on your side let us know about the results

Maybe I can help, but what exactly I need to test?

bjoernQ, 2023-01-13:
A good test would be just trying to connect one of the WiFi examples to an WPA3-only access-point. If it fails probably, it would be helpful to repeat the test with the feature `wifi_logs` and post the logs here

averyanalex, 2023-01-13:
> 

Ok, I will try, but now I can say that with WPA2/WPA3 AP it connects to WPA2.

averyanalex, 2023-01-13:
Logs with WPA2-only and WPA3-only network (SSID: AveryanAlex):
[wpa2.txt](https://github.com/esp-rs/esp-wifi/files/10413088/wpa2.txt)
[wpa3.txt](https://github.com/esp-rs/esp-wifi/files/10413089/wpa3.txt)
As you can see, when network is configured as WPA3-only, AP scan in logs says it is WPA2Personal.

bugadani, 2023-10-26:
This is not the first time I come back to this issue to see if it has some useful info for me but then leave disappointed. Could we edit the title to indicate that this is about an ESP STA trying to connect to a WPA3 access point, and not necessarily WPA3 support in softAP?