esp-rs/esp-wifi-sys #111: Can't build documentation

Fointard, 2023-01-24:
`cargo doc` returns the following error :

```
> cargo doc       
    Updating git repository `https://github.com/esp-rs/esp-pacs`
    Updating crates.io index
    Updating git repository `https://github.com/bjoernQ/bleps`
error: failed to select a version for the requirement `riscv = "^0.9"`
candidate versions found which didn't match: 0.10.1, 0.8.0, 0.7.0, ...
location searched: crates.io index
required by package `riscv-rt v0.10.0`
    ... which satisfies dependency `riscv-rt = "^0.10.0"` of package `esp-wifi v0.1.0 (/home/ash/Rust/esp-wifi)`
```

Is there another way to build the documentation ?

bjoernQ, 2023-01-24:
It would work if you pass the target and the requested features but unfortunately you also run into another issue here:

The currently published HALs for RISCV targets depend on a yanked version of the `riscv` crate which makes it unable to build (unless you happened to build it before that happened and have a `Cargo.lock` file locally). If you are targeting one of the Xtensa chips it should work

If you are targeting a RISCV based chip the bad thing, is you can't just use the HAL as a git dependency (it's fixed there) since there are breaking changes which need to get adapted here (I already started but didn't finish that)

Fointard, 2023-01-24:
It doesn't seem to work with a target either, that's what I get : `error: failed to run `rustc` to learn about target-specific information`
`rustc --print target-list` does not show xtensa targets though, maybe I'm missing sth here.
Does that mean that documentation is currently completely unavailable ? I'm trying to use Wifi on a RISCV-based device...

bjoernQ, 2023-01-24:
Do you set an Xtensa target maybe?

This would create docs
`cargo doc --no-deps --open --target riscv32imc-unknown-none-elf --features "esp32c3,ble,wifi,embedded-svc"`

But as said you will hit the problem of the yanked `riscv` crate.

You could try to place the attached lock file side by side to `Cargo.toml` - not sure if that will work - I never tried it

[Cargo.lock.zip](https://github.com/esp-rs/esp-wifi/files/10491948/Cargo.lock.zip)

Fointard, 2023-01-24:
> 

No luck with that Cargo.lock file either :(

bjoernQ, 2023-01-25:
I hope to get all the changes needed for the upcoming HAL version done and available on a separate branch soon

bjoernQ, 2023-02-01:
We have the updates merged so you should be able to run
`cargo doc --no-deps --open --target riscv32imc-unknown-none-elf --features "esp32c3,ble,wifi,embedded-svc"`

Fointard, 2023-02-01:
> We have the updates merged so you should be able to run `cargo doc --no-deps --open --target riscv32imc-unknown-none-elf --features "esp32c3,ble,wifi,embedded-svc"`

It works, one only need to "+nightly" that cmd of yours :
`cargo "+nightly" doc --no-deps --open --target riscv32imc-unknown-none-elf --features "esp32c3,ble,wifi,embedded-svc"`