esp-rs/esp-wifi-sys #143: Linking/running issue on ESP32

wintermute101, 2023-03-05:
I started simple project on ESP32 when added esp-wifi it stopped linking.
With bunch of "undefined reference to" libc functions like "strncmp" etc.
When I add "-C", "link-arg=-Tesp32_rom_functions.x", to my rustflags it links.
It will not run with error:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7024
0x3fff0030 - _heap_start
    at ??:??
load:0x40078000,len:15400
0x40078000 - __bswapsi2
    at ??:??
load:0x40080400,len:3816
0x40080400 - _init_end
    at ??:??
entry 0x40080648
0x40080648 - ppSearchTxQueue
    at ??:??
I (27) boot: ESP-IDF v5.1-dev-1498-g2d9e75bf05 2nd stage bootloader
I (27) boot: compile time Oct 28 2022 14:39:19
I (29) boot: chip revision: V100
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 003f0000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3ffb0000 size=00a44h (  2628) load
I (113) esp_image: segment 1: paddr=00010a6c vaddr=40080000 size=0422ch ( 16940) load
I (127) esp_image: segment 2: paddr=00014ca0 vaddr=3f404ca0 size=09ab0h ( 39600) map
I (143) esp_image: segment 3: paddr=0001e758 vaddr=4008422c size=02344h (  9028) load
E (147) esp_image: invalid segment length 0xdca1
E (147) boot: Factory app partition is not bootable
E (151) boot: No bootable app partitions in the partition table
ets Jun  8 2016 00:22:57
```

It is simple no_std code, when I tried example from esp-wifi it runs with no issues.
It might be not an issue with esp-wifi but I'm somewhat stuck.

```
[profile.release]
# Explicitly disable LTO which the Xtensa codegen backend has issues
lto = "off"
opt-level = "s"
#strip = "symbols"

[profile.dev]
# Explicitly disable LTO which the Xtensa codegen backend has issues
lto = "off"

[dependencies]
hal = { package = "esp32-hal", version = "0.10.0"}
fugit = "0.3.6"
heapless = "0.7.14"
esp-println = { version = "0.4.0", features = [ "esp32", "log" ] }
esp-backtrace = { version = "0.6.0", features = ["esp32", "panic-handler", "exception-handler", "print-uart"] }
embedded-ads111x = "0.1.0"
esp-wifi = { git = "https://github.com/esp-rs/esp-wifi.git", features = ["esp32", "embedded-svc", "wifi"] }
smoltcp = { version = "0.9.1", default-features=false, features = ["proto-igmp", "proto-ipv4", "socket-tcp", "socket-icmp", "socket-udp", "medium-ethernet", "proto-dhcpv4", "socket-raw", "socket-dhcpv4"] }
embedded-svc = { version = "0.23.1", default-features = false, features = []}
embedded-io = "0.4.0"
toml-cfg = "0.1.3"
```

```
[target.xtensa-esp32-none-elf]
runner = "espflash flash --port /dev/ttyUSB0 --monitor --baud=921600"

rustflags = [
  "-C", "link-arg=-Tlinkall.x",
  "-C", "link-arg=-nostartfiles",
  #"-C", "link-arg=-Tesp32_rom_functions.x",
]

[build]
target = "xtensa-esp32-none-elf"

[unstable]
build-std = ["core"]
```

bjoernQ, 2023-03-05:
Linking the ROM functions is needed as you already figured out - I think the README should mention it. If it doesn't, we need to make that clearer.

The final issue seems to be just a "mis-flash" - it sometimes also happens to me. Probably just doing a clean build and trying to flash again _should_ hopefully fix this

wintermute101, 2023-03-05:
Unfortunately I've tried numerus times doing clean build debug or release with different build options and still the same problem. No luck I would like to get to bottom of this but I'd need some guide to what to try. I  tried changing version of espflash but no luck also.

bjoernQ, 2023-03-06:
Sad to hear that. I just generated a new project from the template and added the dhcp example and it works fine for me on ESP32

[example.zip](https://github.com/esp-rs/esp-wifi/files/10895190/example.zip)

wintermute101, 2023-03-06:
[test.zip](https://github.com/esp-rs/esp-wifi/files/10900057/test.zip)
I've changed some code have not figured out what exactly causes this. This is example that is not working any more.

wintermute101, 2023-03-06:
Ok figured out that adding this:

```rust
    use hal::timer::TimerGroup;
    let timg1 = TimerGroup::new(peripherals.TIMG1, &clocks);
    initialize(timg1.timer0, Rng::new(peripherals.RNG), &clocks).unwrap();
```

To my code fixes the issue. This is weird so basically Rng was added to code?

bjoernQ, 2023-03-07:
Just leaving this here for further investigation.

I can confirm when removing the mentioned lines, I see the same behavior (it would not work without those lines but that is a different thing). Also `esptool` shows the same behavior

Here is a comparison of the "failed" and "fine" builds
```text
FAIL
(bootloader output with esptool)
I (130) esp_image: segment 0: paddr=00010020 vaddr=3ffb0000 size=00080h (   128) load
I (139) esp_image: segment 1: paddr=000100a8 vaddr=3ffb0080 size=009c8h (  2504) load
I (148) esp_image: segment 2: paddr=00010a78 vaddr=40080000 size=00400h (  1024) load
I (156) esp_image: segment 3: paddr=00010e80 vaddr=40080400 size=04d74h ( 19828) load
I (172) esp_image: segment 4: paddr=00015bfc vaddr=40085174 size=0157ch (  5500) load
I (175) esp_image: segment 5: paddr=00017180 vaddr=00000000 size=0dc98h ( 56472)
I (201) esp_image: segment 6: paddr=00024e20 vaddr=3f404e20 size=0c674h ( 50804) map

(bootloader output with espflash)
I (130) esp_image: segment 0: paddr=00010020 vaddr=3ffb0000 size=00a48h (  2632) load
I (140) esp_image: segment 1: paddr=00010a70 vaddr=40080000 size=043a8h ( 17320) load
I (154) esp_image: segment 2: paddr=00014e20 vaddr=3f404e20 size=0c674h ( 50804) map
I (174) esp_image: segment 3: paddr=0002149c vaddr=400843a8 size=02348h (  9032) load

readelf output

There are 39 section headers, starting at offset 0xa5548:

Section Headers:
  [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
  [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] .rtc_fast.text    PROGBITS        400c0000 04b379 000000 00   W  0   0  1
  [ 2] .rtc_fast.dummy   PROGBITS        3ff80000 04b379 000000 00   W  0   0  1
  [ 3] .rtc_fast.data    PROGBITS        3ff80000 04b379 000000 00   W  0   0  1
  [ 4] .rtc_fast.bss     PROGBITS        3ff80000 04b379 000000 00   W  0   0  1
  [ 5] .rtc_fast.noinit  PROGBITS        3ff80000 04b379 000000 00   W  0   0  1
  [ 6] .rtc_slow.text    PROGBITS        50000000 04b379 000000 00   W  0   0  1
  [ 7] .rtc_slow.data    PROGBITS        50000000 04b379 000000 00   W  0   0  1
  [ 8] .rtc_slow.bss     PROGBITS        50000000 04b379 000000 00   W  0   0  1
  [ 9] .rtc_slow.noinit  PROGBITS        50000000 04b379 000000 00   W  0   0  1
  [10] .external.data    PROGBITS        3f800000 04b379 000000 00   W  0   0  1
  [11] .external.bss     PROGBITS        3f800000 04b379 000000 00   W  0   0  1
  [12] .external.noinit  PROGBITS        3f800000 04b379 000000 00   W  0   0  1
  [13] .external_he[...] PROGBITS        3f800000 04b379 000000 00   W  0   0  1
  [14] .rwtext.wifi      PROGBITS        40080400 000400 004d73 00  AX  0   0  4
  [15] .data.wifi        PROGBITS        3ffb0000 006000 000080 00  WA  0   0  4
  [16] .vectors          PROGBITS        40080000 017000 000400 00  AX  0   0  4
  [17] .text             PROGBITS        400d0020 01a020 031359 00  AX  0   0  4
  [18] .rodata           PROGBITS        3f404e20 006e20 00c671 00   A  0   0 16
  [19] .rodata.wifi      PROGBITS        3f411491 013491 00290e 00   A  0   0  1
  [20] .data             PROGBITS        3ffb0080 016080 0009c8 00  WA  0   0  4
  [21] .bss              NOBITS          3ffb0a48 016a48 015569 00  WA  0   0  8
  [22] .noinit           PROGBITS        3ffc5fb4 04b37c 000000 00   W  0   0  4
  [23] .rwtext           PROGBITS        40085174 018174 00157a 00  AX  0   0  4
  [24] .text_heap_start  PROGBITS        400866f0 04b37c 000000 00   W  0   0  4
  [25] .heap_start       PROGBITS        3ffc5fb4 04b37c 000000 00   W  0   0  4
  [26] .comment          PROGBITS        00000000 04b37c 0000ae 01  MS  0   0  1
  [27] .xtensa.info      NOTE            00000000 04b42a 000038 00      0   0  1
  [28] .debug_frame      PROGBITS        00000000 04b464 0018f8 00      0   0  4
  [29] .debug_info       PROGBITS        00000000 04cd5c 015351 00      0   0  1
  [30] .debug_abbrev     PROGBITS        00000000 0620ad 001274 00      0   0  1
  [31] .debug_loc        PROGBITS        00000000 063321 00d1cd 00      0   0  1
  [32] .debug_aranges    PROGBITS        00000000 0704ee 0008a8 00      0   0  1
  [33] .debug_ranges     PROGBITS        00000000 070d96 0025b0 00      0   0  1
  [34] .debug_line       PROGBITS        00000000 073346 009047 00      0   0  1
  [35] .debug_str        PROGBITS        00000000 07c38d 00368b 01  MS  0   0  1
  [36] .symtab           SYMTAB          00000000 07fa18 009520 10     37 297  4
  [37] .strtab           STRTAB          00000000 088f38 01c44f 00      0   0  1
  [38] .shstrtab         STRTAB          00000000 0a5387 0001bf 00      0   0  1
Key to Flags:
  W (write), A (alloc), X (execute), M (merge), S (strings), I (info),
  L (link order), O (extra OS processing required), G (group), T (TLS),
[... 95 lines cut ...]
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  LOAD           0x000400 0x40080400 0x3f400020 0x0974f 0x0974f R E 0x1000
  LOAD           0x00a000 0x3ffb0000 0x3f40976f 0x00190 0x00190 RW  0x1000
  LOAD           0x00a900 0x3f409900 0x3f409900 0x14084 0x14084 R   0x1000
  LOAD           0x01f190 0x3ffb0190 0x3f41d984 0x00f30 0x1b24c RW  0x1000
  LOAD           0x021000 0x40080000 0x40080000 0x00400 0x00400 R E 0x1000
  LOAD           0x021b50 0x40089b50 0x40089b50 0x0164e 0x0164e R E 0x1000
  LOAD           0x024020 0x400d0020 0x400d0020 0x54304 0x54304 R E 0x1000
  GNU_STACK      0x000000 0x00000000 0x00000000 0x00000 0x00000 RW  0x10

[... cut ...] egment mapping:
  Segment Sections...
   00     .rwtext.wifi
   01     .data.wifi
   02     .rodata .rodata.wifi
   03     .data .bss
   04     .vectors
   05     .rwtext
   06     .text .wifiorslpiram.21
   07
```

So, in the failed attempts there is no mapping to `0x400d0020` (text section) but the text section is definitely in the ELF file

@MabezDev @jessebraham any ideas? I still could imagine it's a problem in generation of the image (but then it's present in both `esptool` and `espflash`)

wintermute101, 2023-03-09:
I've been looking at this a little more and there is
[18] .wifiorslpiram.21
section missing all together from non working elf and output image is smaller by 81 kB (sizes of sections are smaller)

MabezDev, 2024-02-20:
This looks to me like an espflash issue. Closing the issue here, feel free to retry and report the issue upstream to espflash.

closed 2024-02-20 by MabezDev