esp-rs/esp-wifi-sys #150: Sending more than buffer size results in infinite loop

teotwaki, 2023-03-24:
Disclaimer: This may be due to user error.

I'm using an ESP32, and I'm able to get connected to wifi and retrieve a DHCP IP. All of that works fine.

I'm trying to transfer a relatively large amount of data (30KB in this case, real use-case could be a few hundred KB up to a few MB). I've devised the following reproduction case:

```rust
#[entry]
fn main() -> ! {
    init_logger(log::LevelFilter::Info);
    init_heap();
    esp_wifi::init_heap();

    println!("Acquiring peripherals");
    let peripherals = Peripherals::take();
    let mut system = peripherals.DPORT.split();
    let clocks = ClockControl::configure(system.clock_control, CpuClock::Clock240MHz).freeze();

    println!("Disabling watchdogs");
    // Disable the RTC and TIMG watchdog timers
    let mut rtc = Rtc::new(peripherals.RTC_CNTL);
    let timer_group0 = TimerGroup::new(peripherals.TIMG0, &clocks);
    let mut wdt0 = timer_group0.wdt;
    let timer_group1 = TimerGroup::new(peripherals.TIMG1, &clocks);
    let mut wdt1 = timer_group1.wdt;

    rtc.rwdt.disable();
    wdt0.disable();
    wdt1.disable();

    let mut socket_set_entries: [SocketStorage; 3] = Default::default();
    let (iface, device, mut controller, sockets) =
        create_network_interface(WifiMode::Sta, &mut socket_set_entries);
    let wifi_stack = WifiStack::new(iface, device, sockets, current_millis);
    esp_wifi::initialize(timer_group1.timer0, rng, &clocks).unwrap();

    let client_config = Configuration::Client(ClientConfiguration {
        ssid: SSID.into(),
        password: PASSWORD.into(),
        ..Default::default()
    });
    let res = controller.set_configuration(&client_config);
    println!("wifi_set_configuration returned {:?}", res);

    controller.start().unwrap();
    println!("is wifi started: {:?}", controller.is_started());

    println!("{:?}", controller.get_capabilities());
    println!("wifi_connect {:?}", controller.connect());

    // wait to get connected
    println!("Wait to get connected");
    loop {
        let res = controller.is_connected();
        match res {
            Ok(connected) => {
                if connected {
                    break;
                }
            }
            Err(err) => {
                println!("{:?}", err);
                loop {}
            }
        }
    }
    println!("{:?}", controller.is_connected());

    // wait for getting an ip address
    println!("Wait to get an ip address");
    loop {
        wifi_stack.work();

        if wifi_stack.is_iface_up() {
            println!("got ip {:?}", wifi_stack.get_ip_info());
            break;
        }
    }

    let mut rx_buffer = [0u8; 1536];
    let mut tx_buffer = [0u8; 1536];
    let mut socket = wifi_stack.get_socket(&mut rx_buffer, &mut tx_buffer);

    println!("Making HTTP request");
    socket.work();

    socket
        .open(Ipv4Address::new(192, 168, 100, 80), 8080)
        .unwrap();

    println!("Sending POST");
    socket
        .write(b"POST /test HTTP/1.0\r\nHost: 192.168.100.80\r\n\r\n")
        .unwrap();

    let v = vec![0; 30000];

    socket.write(&v);

    println!("Finished writing body");
    socket.work();
    socket.flush().unwrap();
    println!("Flushed socket");

    let wait_end = current_millis() + 20 * 1000;
    loop {
        let mut buffer = [0u8; 512];
        if let Ok(len) = socket.read(&mut buffer) {
            let to_print = unsafe { core::str::from_utf8_unchecked(&buffer[..len]) };
            print!("{}", to_print);
        } else {
            break;
        }

        if current_millis() > wait_end {
            println!("Timeout");
            break;
        }
    }
    println!();

    socket.disconnect();

    let wait_end = current_millis() + 5 * 1000;
    while current_millis() < wait_end {
        socket.work();
    }

    loop {}
}
```

Output:

```
I (27) boot: ESP-IDF v5.1-dev-1498-g2d9e75bf05 2nd stage bootloader
I (27) boot: compile time Oct 28 2022 14:39:19
I (29) boot: chip revision: V100
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 003f0000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3ffb0000 size=010b4h (  4276) load
I (114) esp_image: segment 1: paddr=000110dc vaddr=40080000 size=0881ch ( 34844) load
I (135) esp_image: segment 2: paddr=00019900 vaddr=3f409900 size=141a8h ( 82344) map
I (165) esp_image: segment 3: paddr=0002dab0 vaddr=4008881c size=02568h (  9576) load
I (169) esp_image: segment 4: paddr=00030020 vaddr=400d0020 size=55b20h (351008) map
I (297) esp_image: segment 5: paddr=00085b48 vaddr=4008ad84 size=0024ch (   588) load
I (303) boot: Loaded app from partition at offset 0x10000
I (303) boot: Disabling RNG early entropy source...
DSCa�j����alpha starting up...
Acquiring peripherals
Disabling watchdogs
wifi_set_configuration returned Ok(())
is wifi started: Ok(true)
Ok(EnumSet(Client | AccessPoint))
wifi_connect Ok(())
Wait to get connected
Ok(true)
Wait to get an ip address
got ip Ok(IpInfo { ip: 192.168.100.120, subnet: Subnet { gateway: 192.168.100.1, mask: Mask(24) }, dns: Some(192.168.10.5), secondary_dns: None })
Making HTTP request
Sending POST
INFO - not fully written: 1491
INFO - not fully written: 0
INFO - not fully written: 0
INFO - not fully written: 0
INFO - not fully written: 0
INFO - not fully written: 0
INFO - not fully written: 0
INFO - not fully written: 0
INFO - not fully written: 0
INFO - not fully written: 0
INFO - not fully written: 0
...
```

**Please note**: Remember to unplug your ESP32 after running this test; my devboard goes into high-power mode and reaches 60-70°C after a few minutes

My understanding: the code never gets past `socket.write(&v);`. The repeated log seems to come from [here](https://github.com/esp-rs/esp-wifi/blob/main/esp-wifi/src/wifi_interface.rs#L505).

Question: Am I misusing the library, or is this a bug?

bjoernQ, 2023-03-25:
Thanks for reporting this

It's definitely a bug and I think I know how to easily fix it.

A workaround would be to send the data in small chunks until this got fixed

teotwaki, 2023-03-25:
I tried the following approach in `Socket::write`, however this doesn't seem to work out.

```rust
    fn write(&mut self, buf: &[u8]) -> Result<usize, Self::Error> {
        loop {
            self.network.with_mut(|interface, device, sockets| {
                interface.poll(
                    Instant::from_millis((self.network.current_millis_fn)() as i64),
                    device,
                    sockets,
                )
            });

            let (may_send, is_open, can_send) =
                self.network.with_mut(|_interface, _device, sockets| {
                    let socket = sockets.get_mut::<TcpSocket>(self.socket_handle);

                    (socket.may_send(), socket.is_open(), socket.can_send())
                });

            if may_send {
                break;
            }

            if !is_open {
                return Err(IoError::SocketClosed);
            }

            if !can_send {
                return Err(IoError::SocketClosed);
            }
        }

        let mut written = 0;
        loop {
            self.flush()?;

            let res = self.network.with_mut(|_interface, _device, sockets| {
                let socket = sockets.get_mut::<TcpSocket>(self.socket_handle);

                match socket.send_slice(&buf[written..]) {
                    Ok(len) => {
                        written += len;
                        log::info!(
                            "Wrote {} bytes to socket, {} remaining",
                            len,
                            buf.len() - written
                        );
                        Ok(())
                    }
                    Err(err) => return Err(IoError::TcpSendError(err)),
                }
            });

            match res {
                Ok(_) if written < buf.len() => continue,
                Ok(_) => break Ok(written),
                Err(err) => break Err(err),
            }
        }
    }
```

I then also (discarding the code above and going back to `main`) tried shorting out the infinite loop by adding an extra condition:

```rust
                        if written >= buf.len() || !socket.can_send() {
                            break Ok(written);
                        }
```

I then adapted my code to keep writing the buffer with every try. However this ends up with the socket in an invalid state or being dropped (potentially by the HTTP server; unsure).

bjoernQ, 2023-03-25:
Most probably it's just a missing `interface.poll` in that loop

teotwaki, 2023-03-25:
By the way, is there a specific reason you don't call `flush()` in a few places where it appears to be the exact same code block?

teotwaki, 2023-03-25:
The following appears to work. Should I submit a PR?

```rust
    fn write(&mut self, buf: &[u8]) -> Result<usize, Self::Error> {
        loop {
            self.network.with_mut(|interface, device, sockets| {
                interface.poll(
                    Instant::from_millis((self.network.current_millis_fn)() as i64),
                    device,
                    sockets,
                )
            });

            let (may_send, is_open, can_send) =
                self.network.with_mut(|_interface, _device, sockets| {
                    let socket = sockets.get_mut::<TcpSocket>(self.socket_handle);

                    (socket.may_send(), socket.is_open(), socket.can_send())
                });

            if may_send {
                break;
            }

            if !is_open {
                return Err(IoError::SocketClosed);
            }

            if !can_send {
                return Err(IoError::SocketClosed);
            }
        }

        let mut written = 0;
        loop {
            self.flush()?;

            self.network.with_mut(|_interface, _device, sockets| {
                sockets
                    .get_mut::<TcpSocket>(self.socket_handle)
                    .send_slice(&buf[written..])
                    .map(|len| written += len)
                    .map_err(IoError::TcpSendError)
            })?;

            if written >= buf.len() {
                break;
            }
        }

        Ok(written)
    }
```

teotwaki, 2023-03-25:
There is still an odd behaviour here, not sure if related. The buffer seems to be underutilized:

```
2023-03-25T20:25:32.123858Z TRACE hyper::proto::h1::io: received 264 bytes
2023-03-25T20:25:32.134089Z TRACE hyper::proto::h1::io: received 198 bytes
2023-03-25T20:25:32.144833Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.162183Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.172408Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.183498Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.193238Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.204564Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.217121Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.230500Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.242658Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.265793Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.284303Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.301485Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.315141Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.331399Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.358997Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.370164Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.381998Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.393174Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.403077Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.416769Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.433129Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.443791Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.459750Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.475114Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.485273Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.498631Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.514345Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.526147Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.539237Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.551182Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.566878Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.578259Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.588950Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.604090Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.615865Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.626594Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.642510Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.654848Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.672669Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.682211Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.693001Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.703302Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.717173Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.729055Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.742910Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.757375Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.775880Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.786510Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.797735Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.808145Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.819975Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.830790Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.848522Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.858665Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.884687Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.896412Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.909644Z TRACE hyper::proto::h1::io: received 1074 bytes
2023-03-25T20:25:32.921666Z TRACE hyper::proto::h1::io: received 462 bytes
2023-03-25T20:25:32.933178Z TRACE hyper::proto::h1::io: received 659 bytes
```

If I'm not mistaken, 1074 + 462 = 1536. I guess in theory this could be on the receiving side, but somehow I doubt it.

With the code in the reply above, I would expect it to full the tx buffer and then transmit once it's full (through the `flush()`). I'm surprised that the tx buffer keeps being filled (or transmitted) in a flip-flopping size manner.

bjoernQ, 2023-03-26:
To verify if the fragmentation happens on the esp-wifi side you could add a log statement in the `send_data_if_needed` function and log the length of the frame. 

I guess it will fully fill the tx buffer. What you see on the receiving side might be due to fragmentation happening on the router and the receiving host (and any other hops involved - depending on the MTUs configured on every hop in between)

Regarding filing a PR: Sure, seems like a good thing to me

teotwaki, 2023-03-26:
Yeah, it looks like it's on the esp-wifi side:

```
Making HTTP request
INFO - sending... 42 bytes
INFO - sending... 66 bytes
INFO - sending... 54 bytes
Sending POST
Writing content type
INFO - sending... 105 bytes
Writing content length
Writing body
INFO - sending... 267 bytes
INFO - sending... 269 bytes
INFO - sending... 1111 bytes
INFO - sending... 533 bytes
INFO - sending... 1111 bytes
INFO - sending... 533 bytes
INFO - sending... 1111 bytes
INFO - sending... 533 bytes
INFO - sending... 1111 bytes
INFO - sending... 533 bytes
INFO - sending... 1111 bytes
INFO - sending... 533 bytes
INFO - sending... 1111 bytes
INFO - sending... 533 bytes
INFO - sending... 1111 bytes
INFO - sending... 533 bytes
INFO - sending... 1111 bytes
INFO - sending... 533 bytes
Finished writing body
Flushed socket
INFO - sending... 630 bytes
```

Should I create a new issue to track this so we can close this one?

bjoernQ, 2023-03-27:
Yes, please open a separate issue for the observed behavior - probably it's more like a configuration issue internally so it's different from the original issue

teotwaki, 2023-03-27:
This should be closed by #151.

closed 2023-03-27 by teotwaki