esp-rs/esp-wifi-sys #154: Release first version on crates.io

MabezDev, 2023-03-29:
Will fill this issue in later with the required steps/TODOs

jessebraham, 2023-05-23:
Any updates here?

MabezDev, 2023-06-08:
A possible controversial idea before release, remove the `example_util` support crate with the macros. From a maintainability perspective, definitely a lot easier to manage on our side, but makes it quite tricky to copy example code into users real projects.

imo it's worth the extra hassle on our side if it means users can copy and paste more freely.

bjoernQ, 2023-07-05:
In theory we are now ready to release 0.1.0.

Would be good to get #198 in first.

Regarding the examples .... maybe we can also add an example for each chip which just shows the initialization without anything else and mention it in the other examples. The init code hopefully won't change too much so it won't need too much maintenance but we keep the example-utils for the other examples?

MabezDev, 2023-07-05:
I think the biggest problem is that the util stuff is macros instead of normal rust code. What about converting them from macros to functions? At least then users can just copy from there, right now they would have to copy the macros in their project, or unpick the macro logic which is a pain.

bjoernQ, 2023-07-05:
> I think the biggest problem is that the util stuff is macros instead of normal rust code. What about converting them from macros to functions? At least then users can just copy from there, right now they would have to copy the macros in their project, or unpick the macro logic which is a pain.

I think I failed miserably trying to use regular functions for that before coming up with these macros  🤔

DoumanAsh, 2023-07-05:
Could it not be implemented as free functions within corresponding HALs as most of these macros are just `cfg` wrappers?

bjoernQ, 2023-07-06:
Thinking about this again I wonder if changing the examples would really improve the situation for new users at all.

I assume most users will start from something generated by `esp-template`.  That eliminates the need to deal with almost all the macros - the only one left is the one to get the `wifi` or `ble` peripheral via `peripherals.RADIO.split()`

Besides that, the user needs to create/use the right timer - that's it - the remaining code can be copied from any of the examples.

The harder part is to figure out which dependencies and features are needed. Since we are using workspace dependencies and dynamically activate some features that's slightly inconvenient t.b.h.

Last step the user needs to do is remembering to link `rom_functions.x`.

With that in mind I propose this
- keep the examples as they are
- re-export everything from the relevant crates (so the user doesn't need to think about them at all - just depend on esp-wifi with the right features set)
- Change the README chapter `Using in your own binary crate` into something like USERGUIDE.md and link to it
- in the USERGUIDE.md (or whatever we name it) we explain the steps needed to go from a freshly generated project to e.g. being able to scan for access-points - then we explain how to adapt code from the examples

SergioGasquez, 2023-07-06:
> Thinking about this again I wonder if changing the examples would really improve the situation for new users at all.
> 
> I assume most users will start from something generated by `esp-template`. That eliminates the need to deal with almost all the macros - the only one left is the one to get the `wifi` or `ble` peripheral via `peripherals.RADIO.split()`
> 
> Besides that, the user needs to create/use the right timer - that's it - the remaining code can be copied from any of the examples.
> 
> The harder part is to figure out which dependencies and features are needed. Since we are using workspace dependencies and dynamically activate some features that's slightly inconvenient t.b.h.
> 
> Last step the user needs to do is remembering to link `rom_functions.x`.
> 
> With that in mind I propose this
> 
> * keep the examples as they are (maybe add a comment to each example with a link to the USERGUIDE.md - see below)
> * re-export everything from the relevant crates (so the user doesn't need to think about them at all - just depend on esp-wifi with the right features set)
> * Change the README chapter `Using in your own binary crate` into something like USERGUIDE.md and link to it
> * in the USERGUIDE.md (or whatever we name it) we explain the steps needed to go from a freshly generated project to e.g. being able to scan for access-points - then we explain how to adapt code from the examples

A bit off-topic, but maybe we could add a chapter/example on the [no_std training](https://github.com/esp-rs/no_std-training) where we walk the user through all of this

bjoernQ, 2023-07-06:
> A bit off-topic, but maybe we could add a chapter/example on the [no_std training](https://github.com/esp-rs/no_std-training) where we walk the user through all of this.

I like that idea - we can link to that from the README.md, too

MabezDev, 2023-07-25:
> Thinking about this again I wonder if changing the examples would really improve the situation for new users at all.
>
> I assume most users will start from something generated by `esp-template`. That [...]

I like the ideas here, but I'd like to expand on them.

- For `rom_functions.x`, I think we should just move these into esp-hal, the linker won't use the symbols unless they are referenced so there is no downside to doing this.
- Instead of USERGUIDE.md, I think we should
   - Put this in the crate docs
   - Add an option to esp-template to add esp-wifi support
- (Optional) Simplify some of the startup code with some helpers in esp-hal
  - For example, we could easily add a `ClockControl::max(/**/).freeze()` which would ramp the clocks to the highest available on the chip

MabezDev, 2023-08-02:
The `Clock::max` function has been added to esp-hal in https://github.com/esp-rs/esp-hal/pull/701. I also have a proof of concept for the watch dog disable before main: https://github.com/esp-rs/esp-hal/pull/704 which I probably won't finish, myself, but would love for someone to take the reigns on that.

MabezDev, 2023-09-07:
I think we're in a good spot functionality-wise, I think we just need to focus on some documentation (not a tonne, but just bit more) and we should be ready for a crates.io release!

laundmo, 2023-09-07:
i'm trying to use this crate with embassy right now, and i'm stuck setting up dependencies. when cloning this project, examples work fine, but its not very clear from the examples which dependencies a user needs to specify, and which are just brought in by the workspace on accident.

i think some documentation regarding a full project setup with all dependencies, without having to mentally resolve the workspace like i have to for the examples, would help a lot.

MabezDev, 2023-09-07:
> i'm trying to use this crate with embassy right now, and i'm stuck setting up dependencies. when cloning this project, examples work fine, but its not very clear from the examples which dependencies a user needs to specify, and which are just brought in by the workspace on accident.
> 
> i think some documentation regarding a full project setup with all dependencies, without having to mentally resolve the workspace like i have to for the examples, should be one of the documentation goals before releasing.

Imo examples are not the place to describe project setup, even in the embassy org there are bunch of examples that include deps that are not always needed. You can always just take all of the workspace deps and cut them away one-by-one, checking your project still compiles after each one.

The place we need this is inside [esp-template](https://github.com/esp-rs/esp-template). We should add support for embassy projects, and adding esp-wifi.

laundmo, 2023-09-07:
> The place we need this is inside [esp-template](https://github.com/esp-rs/esp-template). We should add support for embassy projects, and adding esp-wifi.

i agree, ideally then the readme should point to that, so that new users dont run into the same issues.

abergmeier, 2023-10-21:
As someone who read all this I am sadly still lost.
My immediate questions are:
1. How should I as a user figure out which deps I need?
2. Even if I have figured out the correct deps - it seems very important to choose the correct features to make it all work together.
3. Is using `esp-idf*` with `esp-wifi` supported?
4. If I encounter errors - should I write Issues or Stackoverflow for these? How are these problems being recorded so that not everyone makes the same mistakes?

jessebraham, 2023-10-22:
It's been about 4 months since we decided that there are no blockers remaining here. Can we at least get the task list updated please so we have some idea of where we stand on this?

bjoernQ, 2023-11-02:
After the first release we probably should maintain a CHANGELOG.md here

jessebraham, 2023-11-14:
Update from the `crates.io` team:

> the crates.io usage policies restrict crates from "using obfuscation to hide or mask functionality". I'm not sure whether pre-compiled closed-source artifacts would fall under that category, but I've put your request on the agenda for our crates.io team meeting on Friday. We'll get back to you once we've made a decision on this.

bjoernQ, 2023-11-27:
After merging #343 we are ready to release