esp-rs/esp-wifi-sys #168: Decide if embedded-svc should automatically get activated with the wifi feature

bjoernQ, 2023-05-03:
See https://github.com/esp-rs/esp-wifi/pull/166#issuecomment-1528863356

Not sure if anyone would like to use the raw unsafe functions but maybe there is a reason to do that?

MabezDev, 2023-05-03:
I'm wondering if we should make it a core dependency. `embedded_svc` aims to be the `embedded_hal` of "services". Also we already use it all over the place, see https://github.com/esp-rs/esp-wifi/blob/48603e220778185d2589172e69fc3c2717f1c20c/esp-wifi/src/wifi/mod.rs#L1399-L1402. If we ripped if we had it optional it would mean duplicating some of its functionality I think.

With that said, if you have some ideas to keep it optional without too much work, I'm open to them!

bjoernQ, 2023-05-04:
After thinking about this a bit, I guess we should include embedded-svc when the wifi feature is active. Since we don't need it for Bluetooth I would still keep the dependency optional but it's probably not worth its own feature

PR #186 (2023-05-25): Miscellaneous Small Things 
This is a collection of very small changes to avoid the overhead of doing full testing each of them

- update _bleps_ dependency
- fixes #170 
- fixes #174 
- fixes #182 
- fixes #168 

Besides of that this makes BLE on ESP32 work again. It's a mystery for me why that broke or why it worked before since I don't think we changes anything interrupt related in esp-hal for months
closed 2023-05-25 by bjoernQ