esp-rs/esp-wifi-sys #172: gettimeofday not implemented

AnthonyGrondin, 2023-05-10:
# Issue

The following function hasn't been implemented yet:
https://github.com/esp-rs/esp-wifi/blob/c446177984b82b5e1658d6b24318e7214460ded8/esp-wifi/src/common_adapter/mod.rs#L428-L431

This makes using `AccessPointConfiguration` with an `auth_method` set to other than `None`, with the `password` field set, panic, when using it in `controller.set_configuration()`

I.E. 
```rust
let ap_config = AccessPointConfiguration {
    ssid: "esp-wifi".into(),
    auth_method: AuthMethod::WPA,
    password: "esp-wifi-password".into(),
    ..Default::default()
};

controller.set_configuration(&ap_config).unwrap();
```

```
!! A panic occured in 'esp-wifi/src/common_adapter/mod.rs', at line 430, column 5
 
PanicInfo {
    payload: Any { .. },
    message: Some(
        not yet implemented: gettimeofday,
    ),
    location: Location {
        file: "esp-wifi/src/common_adapter/mod.rs",
        line: 430,
        col: 5,
    },
    can_unwind: true,
}

```

manio, 2023-08-29:
Today for the first time during my testing I also encounter this **bug**, but it was during long run of my program (it was working fine at the beginning). I just discovered this in the morning after the program was left for night run.

In my case I've got:
```
!! A panic occured in '/home/manio/.cargo/git/checkouts/esp-wifi-836f3b2af57fa847/def22da/esp-wifi/src/common_adapter/mod.rs', at line 430, column 5
 
PanicInfo {
    payload: Any { .. },
    message: Some(
        not yet implemented: gettimeofday,
    ),
    location: Location {
        file: "/home/manio/.cargo/git/checkouts/esp-wifi-836f3b2af57fa847/def22da/esp-wifi/src/common_adapter/mod.rs",
        line: 430,
        col: 5,
    },
    can_unwind: true,
}
 
Backtrace:
 
WARN - esp_wifi_internal_tx 257
WARN - esp_wifi_internal_tx 257
WARN - esp_wifi_internal_tx 257
WARN - esp_wifi_internal_tx 257
WARN - esp_wifi_internal_tx 257
WARN - esp_wifi_internal_tx 257
WARN - esp_wifi_internal_tx 257
WARN - esp_wifi_internal_tx 257
WARN - esp_wifi_internal_tx 257
WARN - esp_wifi_internal_tx 257
WARN - esp_wifi_internal_tx 257
WARN - esp_wifi_internal_tx 257
WARN - esp_wifi_internal_tx 257
WARN - esp_wifi_internal_tx 257
WARN - esp_wifi_internal_tx 257
WARN - esp_wifi_internal_tx 257
WARN - esp_wifi_internal_tx 257
WARN - esp_wifi_internal_tx 257
WARN - esp_wifi_internal_tx 257
WARN - esp_wifi_internal_tx 257
WARN - esp_wifi_internal_tx 257
[...]
```
Unfortunately the backtrace is not there. The same warning line keeps (or was) adding in some specified interval but the ping is not replied back.

bjoernQ, 2023-08-29:
Mhhh this shouldn't happen in normal use (i.e. STA mode). Also, the fact there is no backtrace (it's on an Xtensa target, right? Xtensa should always have a backtrace since it's in their ABI) might indicate that the stack might be corrupted.

The following lines indicate later that trying to send failed a packet failed (makes sense since the wifi task died).

We should certainly implement `gettimeofday` and see if that's already enough to support STA mode with an auth method other than `None` but I guess it's not really the issue in the second message.

manio, 2023-08-29:
Yes, it is ESP32 and STA mode.

PR #304 (2023-10-25): Implement gettimeofday and __assert_func
`__assert_func` or deduplication in `log_timestamp` isn't necessary but wth, being a good boyscout... Closes #172 though it still won't allow my mobile to connect to a WPA2 access point.
closed 2023-10-25 by bjoernQ