esp-rs/esp-wifi-sys #176: Allow changing wifi mode after interface is created.

AnthonyGrondin, 2023-05-15:
# Issue
Currently, the mode of the wifi, is set when the wifi stack is created, and there's no way to change it, after it's initialized.

We should be able to conveniently switch the mode of the interface, between Client and Access Point without having to drop it, and recreate it.

`set_configuration()` seems to already change the wifi mode, but it fails with `InternalError(EspErrWifiMode)`.

https://github.com/esp-rs/esp-wifi/blob/c446177984b82b5e1658d6b24318e7214460ded8/esp-wifi/src/wifi/mod.rs#L1065-L1187

bjoernQ, 2023-05-16:
I guess you need to switch from AP to STA mode?
I don't get that error in my tests but the TCP stack is in a weird state then

AnthonyGrondin, 2023-05-16:
> I guess you need to switch from AP to STA mode? I don't get that error in my tests but the TCP stack is in a weird state then

The error came from calling `controller.connect()` when being in access point mode.

PR #177 (2023-05-17): A way to update iface-config
Fixes #176 

Before a couple of internals were not updated when changing from AP to STA mode or from STA to AP mode. Should hopefully work now

<details>
<summary>Example</summary>

```rust
#![no_std]
#![no_main]
#![feature(c_variadic)]
#![feature(const_mut_refs)]

use examples_util::hal;

use embedded_io::blocking::*;
use embedded_svc::ipv4::{DHCPClientSettings, Interface};
use embedded_svc::wifi::{AccessPointConfiguration, ClientConfiguration, Configuration, Wifi};

use esp_backtrace as _;
use esp_println::logger::init_logger;
use esp_println::{print, println};
u [... cut]
closed 2023-05-17 by bjoernQ