esp-rs/esp-wifi-sys #179: Support Ipv6 for the polling/blocking APIs

bjoernQ, 2023-05-17:
While we already support Ipv6 via embassy-net we don't support it for the polling/blocking API.

When implementing this also consider implementing `embedded-nal` traits.

Another thing to consider: `smoltcp` currently doesn't come with an DhcpV6 implementation so we should implement a DhcpV6 client (probably not in the same PR)

bjoernQ, 2023-05-24:
Basics are done in #181 - further progress seems to be blocked by smoltcp and embassy-net for now

closed 2024-02-20 by MabezDev

urbaniak, 2024-02-20:
are there any news about slaac or dhcpv6? looked a while ago into that and haven't found it in smoltcp

MabezDev, 2024-02-20:
We're using smoltcp, not adding anything on top so I think it would be best to open a issue there :).