esp-rs/esp-wifi-sys #183: Performance issues with wifi initialized

bugadani, 2023-05-24:
This is a bit of a vague issue because I don't have proper data, just what I can see.

I have a device that polls a sensor at 1ms interval, filters, downsamples the signal, and renders it to an SSD1306 (128x64px monochrome) display. This works well enough by itself.

However, when I add esp-wifi and call `esp_wifi::initialize`, even if I do anything more with wifi, the signal rendering becomes very choppy, and it also looks like I'm no longer capturing every sample.

MabezDev, 2023-05-24:
I believe this could (partially) by the task switching frequency, for RISC-V we are using 1000hz switching frequency, but on Xtensa we are using 2000hz for some reason (1/ 500us = 1/ 0.0005 = 2000HZ). At 2000 switches a second, I imagine most of the work done by the CPU is the switching code itself :D.

We should make this the same across all chips and also tune the value. I believe 1000 could still be too high but I guess we need to do some profiling to see.

MabezDev, 2023-05-27:
I suppose this is also fixed by #185 now?

bugadani, 2023-05-27:
I'd say the answer is "yesn't". My particular issue is fixed, and wifi initialization can now be indefinitely delayed, which is good. However, the actualy cause of my problems, that wifi initialization creates two looping tasks without anything to do (ignore their scheduling frequency for now), and that those tasks can't be stopped, still stands. How would you like to track this?

bjoernQ, 2023-05-30:
> I'd say the answer is "yesn't". My particular issue is fixed, and wifi initialization can now be indefinitely delayed, which is good. However, the (original) cause of my problems, that wifi initialization creates two looping tasks without anything to do (ignore their scheduling frequency for now), and that those tasks can't be stopped, still stands. How would you like to track this?

Should be possible to offer a way to completely de-initialize and tear-down the whole WiFi/BLE functionality. Mind creating a new issue for this enhancement?

bugadani, 2023-05-30:
Opened #189.
This issue has been resolved by #185

closed 2023-05-30 by bugadani