esp-rs/esp-wifi-sys #188: promiscuous mode not supported?

gemesa, 2023-05-28:
Hi guys,

I am rather new to the ESP32 world. I would like to ask when will promiscuous mode be supported. I tried to activate it:

```
        esp_wifi_result!(esp_wifi_set_promiscuous(true))?;
        esp_wifi_result!(esp_wifi_set_promiscuous_rx_cb(Some(recv_cb_prom)))?;
        esp_wifi_result!(esp_wifi_set_channel(1, 0))?;
```

But I receive this error:

```
$ screen /dev/ttyUSB0 115200
$ # press EN/RST button
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0048,len:12
ho 0 tail 12 room 4
load:0x3fff0054,len:4800
load:0x40078000,len:17448
load:0x4007c428,len:4840
entry 0x4007c6a0
WiFi started!
              
               
               !! A panic occured in 'esp-wifi/esp-wifi/src/wifi/os_adapter.rs', at line 599, column 5
                                                                                                       
                                                                                                       PanicInfo {
                                                                                                                      payload: Any { .. },
                        message: Some(
                                              not yet implemented: queue_msg_waiting,
                                                                                         ),
                                                                                               location: Location {
                                                                                                                           file: "esp-wifi/esp-wifi/src/wifi/os_adapter.rs",
                                                              line: 599,
                                                                                col: 5,
                                                                                           },
                                                                                                 can_unwind: true,
                                                                                                                  }
                                                                                                                    
                                                                                                                    Backtrace:
         
         0x4008afa8
                   0x401045e4
                             0x400870e0
                                       0x40087312
                                                 0x40083d80
                                                           0x400d8e03
                                                                     0x400d5e04
                                                                               0x40000000
```

As I understood this is the problem:

```
pub unsafe extern "C" fn queue_msg_waiting(_queue: *mut crate::binary::c_types::c_void) -> u32 {
    todo!("queue_msg_waiting")
}
```
There might be more information available in my [sandbox](https://github.com/gemesa/esp32-phantom#promiscuous-mode) if necessary.

bjoernQ, 2023-05-30:
Probably getting promiscuous mode to work will only require this (and maybe a few other) functions to be properly implemented. 

However, we will need a new API to get the raw 802.11 frames since running a network stack on top of this won't make sense

Joe-Austin, 2024-01-26:
Any plans to implement this?

bjoernQ, 2024-01-26:
I think adding support for this would be a good thing, yes - we are just quite busy with other tasks currently but PRs are certainly welcome

Frostie314159, 2024-01-27:
I would be willing to implement these functions, but I may need some help where I should start. We should probably also support sending raw frames.

bjoernQ, 2024-01-29:
Probably we should have another "driver" which allows receiving and sending raw frames - similar to how esp-now is "another" wifi driver: https://github.com/esp-rs/esp-wifi/blob/23b0505445ec337b872b0385d6d49bafcde81071/esp-wifi/examples/esp_now.rs#L38-L39

The additional driver would enable promiscuous mode, set the callback and select the channel

Frostie314159, 2024-01-29:
Ok, seems reasonable. About the functions, @gemesa reported as unimplemented, how would I go about writing the implementations for those?

Joe-Austin, 2024-01-29:
@Frostie314159 That is in `os-adapter`. See https://github.com/esp-rs/esp-wifi/blob/23b0505445ec337b872b0385d6d49bafcde81071/esp-wifi/src/wifi/os_adapter.rs#L531.

I can try to help with this as well if needed, but I am just getting spun up on the ESP ecosystem with Rust.

Frostie314159, 2024-01-30:
Ok. Thanks!

max-niederman, 2024-03-17:
+1 to this. Like @Joe-Austin, I'd also be willing to help but might not be of much use as I'm new to the ecosystem.