esp-rs/esp-wifi-sys #196: Illegal instruction on ESP32C3 after update

katyo, 2023-06-13:
I updated `esp-wifi` in my project which implements BLE gatt server. Now it crashes with exception 'Illegal instruction' at start. Seems `async_ble` example works fine. I double checked used dependencies and features, it is completely same.

Could anybody known that may went wrong?

```
Exception 'Illegal instruction' mepc=0x40385e5c, mtval=0x0000b170
0x40385e5c - ram_bt_set_tx_gain
    at ??:??
0x0000b170 -
    at ??:??
TrapFrame
PC=0x40385e5c         RA/x1=0x40385e54      SP/x2=0x3fc86658      GP/x3=0x3fccea00      TP/x4=0x00000000
0x40385e5c - ram_bt_set_tx_gain
    at ??:??
0x40385e54 - ram_bt_set_tx_gain
    at ??:??
0x3fc86658 - __global_pointer$
    at ??:??
0x3fccea00 - _heap_start
    at ??:??
0x00000000 - os_random
    at /home/bjoern/esp/esp-idf/components/wpa_supplicant/port/os_xtensa.c:48
T0/x5=0x0000005d      T1/x6=0x00000060      T2/x7=0x00006542      S0/FP/x8=0x3fccea90   S1/x9=0x3fc86000
0x0000005d -
    at /home/bjoern/esp/esp-idf/components/wpa_supplicant/src/utils/common.c:563
0x00000060 -
    at /home/bjoern/esp/esp-idf/components/wpa_supplicant/src/utils/common.c:563
0x00006542 -
    at ??:??
0x3fccea90 - _heap_start
    at ??:??
0x3fc86000 - _ZN8esp_wifi6compat12timer_compat6TIMERS17hb3addae244c7180cE
    at ??:??
A0/x10=0x0000002e     A1/x11=0x3fcce9bb     A2/x12=0x3fc8631d     A3/x13=0xfffffffc     A4/x14=0x0000000e
0x0000002e -
    at /home/bjoern/esp/esp-idf/components/wpa_supplicant/port/os_xtensa.c:61
0x3fcce9bb - _heap_start
    at ??:??
0x3fc8631d - phy_param
    at ??:??
0xfffffffc - _rtc_fast_bss_end
    at ??:??
0x0000000e - os_get_time
    at /home/bjoern/esp/esp-idf/components/wpa_supplicant/port/os_xtensa.c:51
A5/x15=0x3fc86000     A6/x16=0x3fc861ea     A7/x17=0x0000000e     S2/x18=0x3fc9c6cc     S3/x19=0x00000000
0x3fc86000 - _ZN8esp_wifi6compat12timer_compat6TIMERS17hb3addae244c7180cE
    at ??:??
0x3fc861ea - phy_param
    at ??:??
0x0000000e - os_get_time
    at /home/bjoern/esp/esp-idf/components/wpa_supplicant/port/os_xtensa.c:51
0x3fc9c6cc - g_phyFuns
    at ??:??
0x00000000 - os_random
    at /home/bjoern/esp/esp-idf/components/wpa_supplicant/port/os_xtensa.c:48
S4/x20=0x3fcceae8     S5/x21=0x3c0468fc     S6/x22=0x00000002     S7/x23=0x00000000     S8/x24=0x00000000
0x3fcceae8 - _heap_start
    at ??:??
0x3c0468fc - _ZN8esp_wifi14common_adapter13phy_init_data21PHY_INIT_DATA_DEFAULT17h7aefdfa83e8ce392E
    at ??:??
0x00000002 - os_random
    at /home/bjoern/esp/esp-idf/components/wpa_supplicant/port/os_xtensa.c:48
0x00000000 - os_random
    at /home/bjoern/esp/esp-idf/components/wpa_supplicant/port/os_xtensa.c:48
0x00000000 - os_random
    at /home/bjoern/esp/esp-idf/components/wpa_supplicant/port/os_xtensa.c:48
S9/x25=0x00000000     S10/x26=0x00000000    S11/x27=0x00000000    T3/x28=0x00000001     T4/x29=0x0000002e
0x00000000 - os_random
    at /home/bjoern/esp/esp-idf/components/wpa_supplicant/port/os_xtensa.c:48
0x00000000 - os_random
    at /home/bjoern/esp/esp-idf/components/wpa_supplicant/port/os_xtensa.c:48
0x00000000 - os_random
    at /home/bjoern/esp/esp-idf/components/wpa_supplicant/port/os_xtensa.c:48
0x00000001 - os_random
    at /home/bjoern/esp/esp-idf/components/wpa_supplicant/port/os_xtensa.c:48
0x0000002e -
    at /home/bjoern/esp/esp-idf/components/wpa_supplicant/port/os_xtensa.c:61
T5/x30=0x00000012     T6/x31=0x00001bf4
0x00000012 - os_get_time
    at /home/bjoern/esp/esp-idf/components/wpa_supplicant/port/os_xtensa.c:51
0x00001bf4 -
    at ??:??

MSTATUS=0x00001801
0x00001801 -
    at ??:??
MCAUSE=0x00000002
0x00000002 - os_random
    at /home/bjoern/esp/esp-idf/components/wpa_supplicant/port/os_xtensa.c:48
MTVAL=0x0000b170
0x0000b170 -
    at ??:??
```

bjoernQ, 2023-06-14:
Seems like the exceptions happens when executing code from IRAM. Probably the memory consumption increased and now the stack grows into code/data

Do you know which revision was working and which revision stopped working?

bjoernQ, 2023-06-23:
You could try https://github.com/esp-rs/esp-hal/pull/616 to see if it is the same problem. 

Unfortunately, it's not enough to just patch the esp-hal dependency since esp-hal now depends on a more recent version of the PAC. But you can use a local copy of esp-wifi and update the dependency there and then patch your dependencies.

Otherwise, we plan to have new releases of esp-hal soon (plus updating esp-wifi)

katyo, 2023-06-23:
@bjoernQ Thanks a lot! Seems issue solved.

closed 2023-06-23 by katyo