esp-rs/esp-wifi-sys #200: esp32 max 1 meter reliable wifi range

karlri, 2023-06-26:
This is awesome! I wish I could dump esp-idf for this with embassy support but I'm having a wifi reliability issue.

esp-idf-hal/esp-idf-svc wifi works without issues with an AP 10 meters away. When I flash the dhcp example of this crate for esp32 on the exact same esp32 unit in the same location, it's almost impossible to get a connection to the AP 10 meters away. I just keep getting error Disconnected. I'm powering the esp32 from usb.

If u use my phone as a hotspot and put the phone right next to the esp32 it does connect and work reliably.

I enabled wifi logs and it looks like that the problem might be related to transmissions never being received by the AP. When I move the phone more than 1m away I start getting lots of log lines about receiving and discarding duplicate packets. I'm thinking that esp32 is probably receiving lots of retransmissions because the AP (phone) is not receiving anything from the esp32. The same behavior is observed with my LAN AP and my phone as AP.  I tried changing wifi_country_t max_tx_power up and down (tested 0 and max legal values) but that seems to have no effect. 

Did anyone else experience connectivity issues on esp32?
Any theories what might be causing this or what I could try?
Could it be a power supply issue in connection with a different power draw profile of this vs esp-idf that could make the supply unstable?

bjoernQ, 2023-06-26:
I have never experienced this but maybe a few questions

- it's the original ESP32, the exact same board works fine with ESP-IDF?
- which WiFi channels are the phone and the LAN AP using? If you can test it: does changing to another channel change anything for you?

karlri, 2023-06-26:
- It's the original esp32 chip, but the board is a cheap $5 one from ebay it's not an original devkit.
- I tried channels 1, 6 and 12 and disabled all power saving measures on the router. Same result. esp-idf-svc wifi connects instantly and reliably even far away. This crate works reliably less than a meter away from the AP, any further and it is almost impossible to connect.

If the esp is more than 2m away from the AP, it can't authenticate. Every other time this happens:
`INFO - auth timeout
INFO - connect status %d -> %d
INFO - stop beacon/connect timer
INFO - reason: auth expire(%d)`

And every other attempt this happens:
`INFO - there is only one ap in blacklist now, release it!
INFO - clear blacklist`

karlri, 2023-06-29:
It seems to be a power issue with my board. I borrowed an oscilloscope and measured the supply voltage and it varies a lot. Probably it gets lower while transmitting. Somehow when my board is connected with usb I have this issue but if I connect a stable power supply to +5v all issues go away.

It seems like the issue with varying voltage is less severe when running esp-idf OS somehow but I'm still closing this because it's very likely a hardware issue and all the boards I ordered have the same issue.

closed 2023-06-29 by karlri

bjoernQ, 2023-06-29:
Thanks for the update on this!