esp-rs/esp-wifi-sys #206: Failed to connected

gho1b, 2023-07-01:
I am trying to use the example codes from the [Embassy DHCP library](https://github.com/esp-rs/esp-wifi/blob/f808488d17abf9c5ac41ddaeb7df65f172e8aad5/examples-esp32s3/examples/embassy_dhcp.rs). However, when I flash my device with the code, it gets stuck at the `Starting WiFi` step. Additionally, after flashing once, I am unable to flash the device again unless I manually erase the flash using the esptool (boot + reset). Can someone please provide clear instructions on how to properly run the example code?

For additional information i use BPI PicoW S3

```
    Finished dev [unoptimized + debuginfo] target(s) in 0.48s
     Running `espflash flash --monitor target\xtensa-esp32s3-none-elf\debug\banana_rs`
[2023-07-01T21:04:04Z INFO ] Serial port: 'COM9'
[2023-07-01T21:04:04Z INFO ] Connecting...
[2023-07-01T21:04:05Z INFO ] Using flash stub
Chip type:         esp32s3 (revision v0.1)
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       48:27:e2:0b:b4:9d
App/part. size:    963,408/8,323,072 bytes, 11.58%
[00:00:00] [========================================]      14/14      0x0
[00:00:00] [========================================]       1/1       0x8000
[00:00:08] [========================================]     510/510     0x10000
[2023-07-01T21:04:15Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378c72
0x40378c72 - __user_exception
    at ~\.cargo\registry\src\index.crates.io-6f17d22bba15001f\esp-backtrace-0.7.0\src\lib.rs:51
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
0x3fce3818 - _stack_end_cpu0
    at ??:??
load:0x403c9700,len:0x4
0x403c9700 - hal_pwr_interrupt_clr_event
    at ??:??
load:0x403c9704,len:0xc00
0x403c9704 - hal_pwr_interrupt_clr_event
    at ??:??
load:0x403cc700,len:0x2eb0
0x403cc700 - hal_pwr_interrupt_clr_event
    at ??:??
SHA-256 comparison failed:
Calculated: b8f485cf73c8ad59b009a332167b4a854ec774585c7fa428926acc30cc0c964b
Expected: 3499e4149a363c2b1ee4ee9709ca5031f50ee6b595a532f697c8659516763ae3
Attempting to boot anyway...
entry 0x403c9908
0x403c9908 - hal_pwr_interrupt_clr_event
    at ??:??
I (45) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (46) boot: compile time Jun  7 2023 08:07:32
I (47) boot: Multicore bootloader
I (51) boot: chip revision: v0.1
I (55) boot.esp32s3: Boot SPI Speed : 40MHz
I (59) boot.esp32s3: SPI Mode       : DIO
I (64) boot.esp32s3: SPI Flash Size : 8MB
I (69) boot: Enabling RNG early entropy source...
I (74) boot: Partition Table:
I (78) boot: ## Label            Usage          Type ST Offset   Length
I (85) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (93) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (100) boot:  2 factory          factory app      00 00 00010000 007f0000
I (108) boot: End of partition table
I (112) esp_image: segment 0: paddr=00010020 vaddr=3c0c0020 size=25e24h (155172) map
I (159) esp_image: segment 1: paddr=00035e4c vaddr=3fc8f420 size=01714h (  5908) load
I (161) esp_image: segment 2: paddr=00037568 vaddr=3fcb141c size=002c4h (   708) load
I (165) esp_image: segment 3: paddr=00037834 vaddr=40378000 size=07424h ( 29732) load
I (182) esp_image: segment 4: paddr=0003ec60 vaddr=00000000 size=013b8h (  5048) 
I (184) esp_image: segment 5: paddr=00040020 vaddr=42000020 size=bb30ch (766732) map
I (383) boot: Loaded app from partition at offset 0x10000
I (384) boot: Disabling RNG early entropy source...
INFO - coex firmware version: %s

INFO - coexist rom version %s

INFO - pp rom version: %s

INFO - net80211 rom version: %s

INFO - wifi driver task: %x, prio:%d, stack:%d, core=%d
INFO - wifi firmware version: %s
INFO - wifi certification version: v%d.%d
INFO - config NVS flash: %s
INFO - config nano formating: %s
INFO - Init data frame dynamic rx buffer num: %d
INFO - Init management frame dynamic rx buffer num: %d
INFO - Init management short buffer num: %d
INFO - Init dynamic tx buffer num: %d
INFO - Init static tx FG buffer num: %d
INFO - Init static rx buffer size: %d
INFO - Init static rx buffer num: %d
INFO - Init dynamic rx buffer num: %d
INFO - hint=%d, act_dur=<%d,%d>, pas_dur=%d, nchans=%d history_num=%d
INFO - @WIFIFRAME [ff, ff, ff, ff, ff, ff, 48, 27, e2, 0b, b3, 9c, 08, 00, 45, 00, 01, 22, 00, 00, 40, 00, 40, 11, 39, cc, 00, 00, 00, 00, ff, ff, f
f, ff, 00, 44, 00, 43, 01, 0e, 85, 34, 01, 01, 06, 00, 07, ef, 3b, d9, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 0
0, 48, 27, e2, 0b, b3, 9c, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 0
0, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 0
0, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 0
0, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 0
0, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 0
0, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 63, 82, 53, 63, 35, 01, 01, 3d, 07, 01, 48, 27, e2, 0
b, b3, 9c, 39, 02, 05, 82, 37, 03, 01, 03, 06, ff]
WARN - esp_wifi_internal_tx 12290
WARN - Unknown wifi mode in link_state
start connection task
Device capabilities: Ok(EnumSet(Client | AccessPoint))
INFO - clear blacklist
Starting wifi
```

MabezDev, 2023-07-01:
To eliminate any network condition issues, can you try connecting with an esp-idf example?

gho1b, 2023-07-02:
> To eliminate any network condition issues, can you try connecting with an esp-idf example?

I can run the idf intro example (std) without any error and success to connect in my network and get the ip address. I also try to run the dhcp example using `cargo run --example dhcp --release --features "embedded-svc,wifi"` and got stuck in `wifi_set_configuration returned Ok(())` step as following terminal result
```
[2023-07-02T10:56:43Z INFO ] Serial port: '/dev/ttyACM0'
[2023-07-02T10:56:43Z INFO ] Connecting...
[2023-07-02T10:56:43Z INFO ] Using flash stub
Chip type:         esp32s3 (revision v0.1)
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       48:27:e2:0b:b4:9d
App/part. size:    477,088/8,323,072 bytes, 5.73%
[00:00:00] [========================================]      14/14      0x0                                                                                                                                                                                                                                           [00:00:00] [========================================]       1/1       0x8000                                                                                                                                                                                                                                        [00:00:04] [========================================]     285/285     0x10000                                                                                                                                                                                                                                       [2023-07-02T10:56:49Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378c72
0x40378c72 - Timer2
    at ??:??
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
0x3fce3818 - _stack_end_cpu0
    at ??:??
load:0x403c9700,len:0x4
0x403c9700 - hal_pwr_interrupt_clr_event
    at ??:??
load:0x403c9704,len:0xc00
0x403c9704 - hal_pwr_interrupt_clr_event
    at ??:??
load:0x403cc700,len:0x2eb0
0x403cc700 - hal_pwr_interrupt_clr_event
    at ??:??
SHA-256 comparison failed:
Calculated: b8f485cf73c8ad59b009a332167b4a854ec774585c7fa428926acc30cc0c964b
Expected: 3499e4149a363c2b1ee4ee9709ca5031f50ee6b595a532f697c8659516763ae3
Attempting to boot anyway...
entry 0x403c9908
0x403c9908 - hal_pwr_interrupt_clr_event
    at ??:??
I (45) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (46) boot: compile time Jun  7 2023 08:07:32
I (47) boot: Multicore bootloader
I (51) boot: chip revision: v0.1
I (55) boot.esp32s3: Boot SPI Speed : 40MHz
I (59) boot.esp32s3: SPI Mode       : DIO
I (64) boot.esp32s3: SPI Flash Size : 8MB
I (69) boot: Enabling RNG early entropy source...
I (74) boot: Partition Table:
I (78) boot: ## Label            Usage          Type ST Offset   Length
I (85) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (93) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (100) boot:  2 factory          factory app      00 00 00010000 007f0000
I (108) boot: End of partition table
I (112) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=127b0h ( 75696) map
I (139) esp_image: segment 1: paddr=000227d8 vaddr=3fc8d320 size=01200h (  4608) load
I (141) esp_image: segment 2: paddr=000239e0 vaddr=3fca9164 size=00168h (   360) load
I (145) esp_image: segment 3: paddr=00023b50 vaddr=40378000 size=0531ch ( 21276) load
I (160) esp_image: segment 4: paddr=00028e74 vaddr=00000000 size=071a4h ( 29092) 
I (169) esp_image: segment 5: paddr=00030020 vaddr=42000020 size=54758h (345944) map
I (258) boot: Loaded app from partition at offset 0x10000
I (259) boot: Disabling RNG early entropy source...
wifi_set_configuration returned Ok(())
```
thats mean the example always stuck when try to call `controller.start().unwrap();` i was build in windows 10 64bit and ubuntu jammy 22. Both os has same result
[... 109 lines cut ...]
I (713) banana_rs_idf::wifi: Scanning...
I (5813) banana_rs_idf::wifi: Found configured access point MYNETWORK on channel 3
I (5813) wifi:mode : sta (48:27:e2:0b:b4:9d) + softAP (48:27:e2:0b:b4:9e)
I (5813) wifi:Total power save buffer number: 16
I (5823) wifi:Init max length of beacon: 752/752
I (5823) wifi:Init max length of beacon: 752/752
I (5833) wifi:Total power save buffer number: 16
I (5833) banana_rs_idf::wifi: Connecting wifi...
I (8673) wifi:new:<3,1>, old:<3,1>, ap:<3,1>, sta:<3,0>, prof:3
I (9723) wifi:state: init -> auth (b0)
I (9723) wifi:state: auth -> assoc (0)
I (9723) wifi:state: assoc -> run (10)
I (9833) wifi:connected with MYNETWORK, aid = 3, channel 3, BW20, bssid = 0c:37:47:94:04:a1
I (9833) wifi:security: WPA2-PSK, phy: bgn, rssi: -67
I (9833) wifi:pm start, type: 1

I (9833) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
I (9853) wifi:<ba-add>idx:0 (ifx:0, 0c:37:47:94:04:a1), tid:6, ssn:5, winSize:64
I (9853) banana_rs_idf::wifi: Waiting for DHCP lease...
I (9883) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (11943) wifi:<ba-add>idx:1 (ifx:0, 0c:37:47:94:04:a1), tid:0, ssn:0, winSize:64
I (12853) esp_netif_handlers: sta ip: 192.168.1.11, mask: 255.255.255.0, gw: 192.168.1.1
I (12853) banana_rs_idf::wifi: Wifi DHCP info: IpInfo { ip: 192.168.1.11, subnet: Subnet { gateway: 192.168.1.1, mask: Mask(24) }, dns: Some(1.1.1.1), secondary_dns: Some(1.0.0.1) }
I (13863) banana_rs_idf: Hello, world!
```

bjoernQ, 2023-07-02:
If your board is connected via the Serial/JTAG (your comment about not being able to flash it again without manually entering download mode suggests that) you need the `phy-enable-usb` feature since otherwise your USB connection dies when the phy is initialized (so maybe it doesn't fail to connect but you just don't get any output)

gho1b, 2023-07-02:
> phy-enable-usb

Got it, thank you. i thought i should be close.

closed 2023-07-02 by gho1b