esp-rs/esp-wifi-sys #211: `WifiController::stop` hangs and kills debug session

bugadani, 2023-07-14:
I have a bizarre issue where I'm trying to deinitialize wifi AP (before restarting in STA mode). I'm stopping the embassy tasks that run the stack and handle the wifi events (this most likely drops a `wait_for_event` future). Then, I'm calling `stop` on the controller, at which point my MCU disconnects from the debugger and the USB serial, and locks up. At this point I need to short EN to GND to resurrect the MCU.

I'm assuming I need to call `stop` because the idf docs suggest that it frees up memory allocated by the AP mode. In the past, I've used `esp_wifi_set_mode` to switch between AP/STA, which I don't think is available through esp-wifi. I've used `esp_wifi_stop` to stop STA mode previously on my esp-idf-based project, and I believe I've seen `stop` work here as well to stop a STA controller, but take this last one as anecdata.

I'm wondering if `esp_wifi_stop` assumes something that the current implementation does not provide.

MabezDev, 2023-07-14:
You're debugging via USB-SERIAL-JTAG right? It's possible that calling `stop` will also turn off the clocks that are also powering the USB-SERIAL-JTAG, after all, we have a feature specifically to keep the clock on :thinking:.

bugadani, 2023-07-15:
What bugs me is that I don't think stopping STA has the same issue. I will have to investigate more, though because so far I've mainly used AP mode and didn't pay much attention to STA.

bjoernQ, 2023-07-17:
> I'm wondering if esp_wifi_stop assumes something that the current implementation does not provide.

There are a few things unimplemented but in general calling these unimplemented functions should panic (or we _know_ it's okay to have an empty stub there). Certainly, there might be something that slipped through

bugadani, 2023-07-19:
I tried connecting an external debugger but all I'm getting is nonsense: call stack indicates my app is in one state (that it absolutely should not be in!) on the bottom, then a few levels up it's polling the future that is stopping the AP.

Either something is very wrong here and stopping the AP completely messes up my firmware, or I am not able to reliably debug my application after the code hits `stop`. Maybe the two are related, I have no idea.

bjoernQ, 2023-07-19:
Mhhh - so the way to reproduce this is to start an AP and then stop it?
And that works fine with STA?

bugadani, 2023-07-19:
I should probably minimize a repro, but yes, it seems that seems to be the case - stopping a WifiController that was returned by `esp_wifi::wifi::new_with_mode(&init, wifi, WifiMode::Ap);`.

closed 2023-08-07 by bjoernQ