esp-rs/esp-wifi-sys #212: No DNS Socket interface

Sofiman, 2023-07-17:
Hey, 
I am trying to do DNS queries to get IP addresses from a hostname. It seems that there is no method on WifiStruct nor a DNSSocket struct in the wifi_interface.rs file to create a DNS socket. This limits an implementation of a fully featured HTTP client without specifying explicit IP addresses every time. I looked up the source code of WifiStack and how Socket and UDPSocket, and I could not figure out the wrapping of the smoltcp socket types. It is especially frustrating when trying to utilize the full smoltcp features through esp-wifi. I'm using an esp32-s3 with no-std, and I would not mind contributing to bring the DNSSocket to esp-wifi if needed.

bjoernQ, 2023-07-18:
It should be possible to use the raw smoltcp instead of our wrapper if you like (to be honest I haven't tried that for a long time - maybe there is something preventing that currently). The wrapper is really just meant for convenience.

For async you should be able to use DNS via embassy-net already. For sync we ideally would add something like https://docs.rs/embassy-net/0.1.0/embassy_net/struct.Stack.html#method.dns_query (but not async obviously)

Sofiman, 2023-07-18:
Got it, I think indeed that a dns_query method on WifiStack is more convenient than a whole DNSSocket struct. I'll try to open an PR if I manage to get it working. Thanks for the reply!

closed 2023-07-20 by Sofiman