esp-rs/esp-wifi-sys #214: wifi init cause issue on esp32

alexiionescu, 2023-07-19:
I have problem with a project on esp32 xtensa using `esp-wifi` (with `esp-now`) and `embassy` `async`. I have tried using xtensa rust 1.69 and latest 1.71 with the same outcome.

1. When I run only `esp-wifi` (with `esp-now`) the firmware hangs after about 12 hours. I need to reset to recover. This exact firmware works without issue on esp32c3 for weeks.
The amount of data traffic on the `esp-now` is minimal (a keep-alive packet received every 3 minutes, one sent every 5 minutes).
2. When run with another `embassy` task that reads `async` from an uart I have 2 outcomes:

- firmware hangs in about 20-30 minutes without errors on uart
- firmware start to give me corrupt data on Uart read after 20-30 minutes and hangs in about 1 hour (memory corruption?). I have analyzed the corruption and some bytes are basically corrupted. Length seems to be fine.

If I run the Uart task alone reads without error and never hangs (I have leave it running about 24 hours).
The Uart traffic is significant (I have a constant 5 KB/s data read at baud rate 921600)

The issue with Uart reproduces if only calling `esp-wifi::initialize` and nothing else. I have tried several times and reproduces within 10-40 minutes.

So my current plan is to drive a local copy of tsp-wifi and comment stuff from `esp-wifi::initialize` until the issue is fixed.
Then report back to you guys. If you have any advice or a better approach please let me know.

P.S> I have ask this question on the esp-rs room, but I guess is better to have it here.

alexiionescu, 2023-07-20:
> First idea would be that it's related to leaking memory or excess stack usage. After `initialize` the wifi task is running so one idea would be to see if changing the task's stack size in
> 
> https://github.com/esp-rs/esp-wifi/blob/68dc11bbb2c0efa29c4acbbf134d6f142441065e/esp-wifi/src/preempt/mod.rs#L29
> 
> changes anything. (i.e. increasing it slightly extends the time until things go belly-up)
> If that doesn't change anything it would be interesting if it's running out of heap memory

I tried from small increase (10240) up to 4x increase(32768), it does not change the time to failure. 
How can I check if is running out of heap memory?

bjoernQ, 2023-07-20:
We already log a message at level `warn` here: https://github.com/esp-rs/esp-wifi/blob/37dd9ed7ae07a86f7afaef34fa4cbb48ad41b210/esp-wifi/src/compat/malloc.rs#L19-L22

when we are unable to allocate memory which you should be able to see if logging is enabled. 

Unfortunately, it's a normal thing and not necessarily a hard error if that happens (and it definitely happens). The calling code is supposed to handle such a situation gracefully.

On the other hand, reading your issue again it might be the case that for some reason the main-task's stack grows into the data/bss segment and overwrites data (especially seeing UART start sending garbage - is whatever is sent via UART from a statically allocated buffer?). However, if you don't do anything but initializing WiFi there is also nothing esp-wifi related running in the main task. Also, interesting this doesn't happen on ESP32-C3

alexiionescu, 2023-07-20:
> We already log a message at level `warn` here:
>
> https://github.com/esp-rs/esp-wifi/blob/37dd9ed7ae07a86f7afaef34fa4cbb48ad41b210/esp-wifi/src/compat/malloc.rs#L19-L22
>
> when we are unable to allocate [...]

The UART1 task is only receiving from a slave board in a static stack allocated buffer and not sending anything. The received data is corrupted after a certain amount of time, so something overflows and start writing over the stack I guess.
The UART0 is sending esp-println data to console. I have disabled the log crate because it was not adding \r in esp-println and on apps like screen shows terrible. Made my own logging. 

Anyway, I have put the logging back now, as a feature. It does not show that warning at all. I have run it with the original task_stack (8192). Another interesting fact is that when corruption occurs the hangs occurs later. Also the corrupt number of bytes grows over time. I guess if it overflows on uart1 buffer it has some more space until overflows on something even more important and everything halts. When this does not happen it overflows to the important part and halts much quicker.

I do not have the UART1 task on ESP32-C3 because this task runs only on this special hardware.
What I have run is only Esp-Now task: on ESP32-C3 works without issue, on this special hardware (without UART1 task) hangs in about 12 hours. 

the uart1 task is an `embassy_executor::task` that looks like below. As mentioned, without wi-fi init works no issues for a long time, no data corruption.
```rust
pub const UART_READ_BUF_SIZE: usize = 0x60;

#[embassy_executor::task]
pub async fn run(mut uart: Uart<'static, UART1>) {
    const MAX_BUFFER_SIZE: usize = 20 * UART_READ_BUF_SIZE;
    const MOVE_BUFFER_SIZE: usize = MAX_BUFFER_SIZE - 3 * UART_READ_BUF_SIZE;
    // the stack buffer for receive
    let mut rbuf = [0u8; MAX_BUFFER_SIZE];
    let mut rbuf_len = 0;
    let mut rbuf_ofs = 0;
    loop {
       match uart.read(&mut rbuf[rbuf_len..]).await {
            Ok(len) if len > 0 => {
                rbuf_len += len;
                //process data &rbuf[rbuf_ofs ..rbuf_len], increase rbuf_ofs 
                if rbuf_ofs == rbuf_len {
                    rbuf_ofs = 0;
                    rbuf_len = 0;
                } else if rbuf_len > MOVE_BUFFER_SIZE && rbuf_ofs > 0 {
                    rbuf.copy_within(rbuf_ofs..rbuf_len, 0);
                    rbuf_len -= rbuf_ofs;
                    rbuf_ofs = 0;
                }
          }
          ... //errors, none occurs
      }
}
```

bjoernQ, 2023-07-21:
The posted code looks sane to me

Probably would be best if you could create a public repro with the minimal amount of code to reproduce this issue

alexiionescu, 2023-07-21:
> The posted code looks sane to me
> 
> Probably would be best if you could create a public repro with the minimal amount of code to reproduce this issue

Published here: [esp32-wifi-uart](https://github.com/alexiionescu/esp32-wifi-uart.git). Issue start reproducing after about 30 minutes (data corruption) and halts after 1h.
For sanity check,I have run with `esp_wifi::initialize` call commented and verified no issue occurs.

Problem with this code is that requires a special hardware to reproduce. And a lot of BLE devices nearby to reproduce faster.
The hardware is this: [MKGW-mini](https://www.mokosmart.com/moko-mini-ble-to-wifi-gateway-data-relay-between-beacons-and-your-cloud-server)

bjoernQ, 2023-07-21:
Thanks for the code to reproduce the issue. I tried reproducing it by running the code on ESP32 and have my PC sending `Hello World!` to it via UART1. I had it running for more than 1h and it didn't crash or halt for me.

Since I don't send valid data it never gets into the parsing code for me but from looking at the code there is nothing suspicious happening there which could explain the behavior you see.

alexiionescu, 2023-07-21:
> Thanks for the code to reproduce the issue. I tried reproducing it by running the code on ESP32 and have my PC sending `Hello World!` to it via UART1 in a loop. I had it running for definitely more than 1h and it didn't crash or halt for me.
> 
> Since I don't send valid data it never gets into the parsing code for me but from looking at the code there is nothing suspicious happening there which could explain the behavior you see.

Thank you very much for looking into this.

I have just took the esp32 at my house and run the same test. Worked without issue for more than 1h  too.
There are 2 differences:
1. The number of part message is 3 times less often(less BLE packets in the air). I will wait and see if the issue appears after more time has passed.
2. At the office I have 2 esp32c3 communicating asp-now. I have also noticed that when I took 1 esp32c3 and move it further away the timing of the reproduction changed(increased). So a question will be, if there are other esp-now packets in the air that do not belong to the esp32 (p2p between the 2 esp32c3) is the stack getting them and store them anywhere in the memory?

But at this point I am suspecting #2. I will let it run for the night to make sure is not #1, and let you know the result.

P.S> The art protocol is very simple: 0xED 0x02 0x04 N(1 byte) (N+8) bytes ...

alexiionescu, 2023-07-22:
> Thank you very much for looking into this.
>
> I have just took the esp32 at my house and run the same test. Worked without issue for more than 1h too. There are 2 differences:
>
> 1. The number of part [...]

Update: it is no.1. It happens, just 3 times slower, proportional with the amount of data. But does not halt even after 5h, only data corruption.
Is like the time to occur depends on the `embassy` wakes count. 
I will make a simple ticker, every 5ms, no uart, see if it still happens.

alexiionescu, 2023-07-22:
> Update: it is no.1. It happens, just 3 times slower, proportional with the amount of data. But does not halt even after 5h, only data corruption. Is like the time to occur depends on the `embassy` [...]

There is no issue on using only a ticker. So only on UART and depends on the amount of received data.

alexiionescu, 2023-07-26:
I have update the sample on git with a version using only `esp_now` (use feature `esp_now` )
I have run on an esp32 and esp32c3 in parallel running this code. The esp32 stopped working in about 1h. 
There were other esp32c3 transmitters in the air and this could have contributed to the faster reproduction.

P.S> The `esp-wifi` and `esp-hal` (latest git version that contains my UART PR fix) are not compatible anymore. The esp_now::init hangs for esp32 and for esp32c3 does not even compile anymore(some duplicate symbols on linking). So using uart is not an option to reproduce. I have also deleted my own PR branch (based on git suggestion !!!).

bjoernQ, 2023-07-26:
Thanks for the update. I am currently on vacation and won't be really able to get to this.
One thing which might be worth to try (but it's quite some effort) would be to test the last esp-wifi commit before the drivers were updates (should be abf6d4b781675a90e3ca7bba467c0bf9e4710488 ) - I think you need to downgrade the esp-hal dependency for that. 

Just to check if the latest drivers introduced the problems seen here

bjoernQ, 2023-08-07:
Ok just to double check. 

esp-hal 8598e188b1d1c2d9d7fe2eccaf136bdb693f490b was the last known working commit.
The first non-working commit is debe2b80048e00ba63a36e4dddc844fa128a9173

There are no obviously dangerous changes in that range so probably a `git bisect` would be the best next step

alexiionescu, 2023-08-07:
> Ok just to double check.
> 
> esp-hal 8598e188b1d1c2d9d7fe2eccaf136bdb693f490b was the last known working commit. The first non-working commit is debe2b80048e00ba63a36e4dddc844fa128a9173
> 
> There are no obviously dangerous changes in that range so probably a `git bisect` would be the best next step
I have run git bisect
- for esp32c3 linking issue I got this commit
    [213dde9304ae5d6aa23def2c1703520d4683ae52] Add a new RMT driver (#653)
- for the `esp_now::EspNow::init` lock on esp32 (internally is `esp_wifi_start()` that locks) I got
    [9f5e2b59de752b24c93ab96f262104a27a154c06] ESP32-S3 deep sleep (#660)

bugadani, 2023-08-07:
> for the esp_now::EspNow::init lock on esp32 (internally is esp_wifi_start() that locks) I got
[9f5e2b59de752b24c93ab96f262104a27a154c06] ESP32-S3 deep sleep (#660)

This is a known issue (sorry about it!) that should be fixed by https://github.com/esp-rs/esp-wifi/pull/220

alexiionescu, 2023-08-07:
> > for the esp_now::EspNow::init lock on esp32 (internally is esp_wifi_start() that locks) I got
> > [9f5e2b59de752b24c93ab96f262104a27a154c06] ESP32-S3 deep sleep (#660)
> 
> This is a known issue (sorry about it!) that should be fixed by #220. I probably should not have messed with ESP32 code there, reordering caused wifi clocks to become disabled on startup.

Thanks,
working now.

alexiionescu, 2023-08-11:
> * for esp32c3 linking issue I got this commit
>   [213dde9304ae5d6aa23def2c1703520d4683ae52] Add a new RMT driver (#653)

This is fixed now in esp-hal 0.11.0 release.

Now we are back on the initial issue, which is still not working on esp32 chipset. Maybe this hardware I am has a problem.
Could the version of esp32 matter? This hardware uses an older version ESP32­WROOM­32 which is not recommended for new designs according to espressif site.

Meanwhile, I have also tried running esp-wifi without async mode and the problem is the same. However the uart part was still embassy async, only asp-wifi was without async.

bugadani, 2023-08-11:
Could you please try a `cargo update` to make sure that you're using `embassy-executor = 0.2.1`? That release includes a fix for hanging ESP32 firmware that may or may not causes your issue.

bugadani, 2023-08-11:
> > Could you please try a `cargo update` to make sure that you're using `embassy-executor = 0.2.1`? That release includes a fix for hanging ESP32 firmware that may or may not causes your issue.
> 
> Already did that based on the #226 comments.
> With the new embassy version is running for almost 3h now, which is longer than any time so far. I will let it run over the weekend and see how it goes.
> However this does not solves my UART data corruption. It is still there. But maybe, the two issues are unrelated. At least we can treat them separately.

To clarify, it is still running after 3h, do I read you correctly?

alexiionescu, 2023-08-14:
> > > > Could you please try a `cargo update` to make sure that you're using `embassy-executor = 0.2.1`? That release includes a fix for hanging ESP32 firmware that may or may not causes your issue.
> > > 
> > > 
> > > Already did that based on the #226 comments.
> > > With the new embassy version is running for almost 3h now, which is longer than any time so far. I will let it run over the weekend and see how it goes.
> > > However this does not solves my UART data corruption. It is still there. But maybe, the two issues are unrelated. At least we can treat them separately.
> > 
> > 
> > To clarify, it is still running after 3h, do I read you correctly?
> 
> Correct, now almost 4h and still running. Seems that freeze is solved by latest embassy(hopefully it will not prolong the time until the freeze appears, but I will let it run all the weekend). Not sure why this happens on esp32 and not esp32c3.

I can confirm the freeze problem is fixed. Has run for more than 48h.

alexiionescu, 2023-08-14:
The only issue remaining is the UART1 data corruption in the presence of `esp-wifi`.
- If `esp_wifi::initialize` is not called it works without any issue.
- if `esp_wifi::initialize` is called it works with few corruption once in a while.
- if `esp-now` is fully functional the corruption gets worst. But is intermittent, for example I can found 30 minutes of continuous running without any errors, then errors appear again, and so on.

bjoernQ, 2023-08-14:
> The only issue remaining is the UART1 data corruption in the presence of `esp-wifi`.
> 
> * If `esp_wifi::initialize` is not called it works without any issue.
> * if `esp_wifi::initialize` is called it works with few corruption once in a while.
> * if `esp-now` is fully functional the corruption gets worst. But is intermittent, for example I can found 30 minutes of continuous running without any errors, then errors appear again, and so on.
> 
> Could `esp-wifi` interfere with the RX Fifo RAM Mapped Memory?

The big difference after executing `esp_wifi::initialize` is that the scheduler runs three tasks in total (main, wifi-task, wifi-timer-task). The scheduler uses a tick of 100Hz by default - so worst case the main task won't progress for 20ms. 

Given the UART driver is CPU only that's long enough to overrun the FIFO at a high baud-rate. 

When nothing really happens in esp-wifi the wifi taks should quickly yield and the effect should be less likely.  (Which matches your observations I guess)

If that's really the cause
- lowering the baud-rate should lower the likeliness of errors
- changing the scheduler's tick frequency should change the likeliness of errors

bjoernQ, 2023-08-14:
Yes, I guess an error should get triggered.

> Lowering the baud rate is not an option for me, it will required overwrite the nRF slave board firmware which would require a lot of work for me.

That wouldn't be a solution - I just meant that as a test to see if it changes anything. Maybe you could simulate the nRF side on a PC

> Scheduler's tick frequency: I am already using esp-hal-common/esp32_40mhz. Is this the setting or Is there another option?

We now have a way to configure it since 87687c82db75bb7378957c48d990602f423c7d36 

See https://github.com/esp-rs/esp-wifi/blob/main/docs/tuning.md - the config option is `tick_rate_hz`

alexiionescu, 2023-08-14:
> Yes, I guess an error should get triggered.
> 
> > Lowering the baud rate is not an option for me, it will required overwrite the nRF slave board firmware which would require a lot of work for me.
> 
> That wouldn't be a solution - I just meant that as a test to see if it changes anything. Maybe you could simulate the nRF side on a PC
> 
> > Scheduler's tick frequency: I am already using esp-hal-common/esp32_40mhz. Is this the setting or Is there another option?
> 
> We now have a way to configure it since [87687c8](https://github.com/esp-rs/esp-wifi/commit/87687c82db75bb7378957c48d990602f423c7d36)
> 
> See https://github.com/esp-rs/esp-wifi/blob/main/docs/tuning.md - the config option is `tick_rate_hz`

I noticed now that the bps received is less when `esp-wifi` is on. 
So I loose bytes, even the error is not triggered, maybe the system is loosing them somehow because CPU is busy doing something else?
I will try the setting and see if that changes anything. I did not knew about that toml-cfg, I will use it for my code too, looks nice.

manio, 2023-08-14:
@alexiionescu maybe you could also think about handling the data in the interrupt?
https://github.com/esp-rs/esp-hal/blob/main/esp32-hal/examples/serial_interrupts.rs

alexiionescu, 2023-08-14:
> @alexiionescu maybe you could also think about handling the data in the interrupt?
> https://github.com/esp-rs/esp-hal/blob/main/esp32-hal/examples/serial_interrupts.rs

I will try that too, thanks.

alexiionescu, 2023-08-17:
> I don't know your code but this is all really strange. The code under `#[interrupt]` is done at the highest priority in a sense that any other task cannot interrupt this execution. Keep in mind that this code has to be as short as possible - i mean eg. collect data but leave the processing in some other "normal" task.

My code is here published here: [esp32-wifi-uart](https://github.com/alexiionescu/esp32-wifi-uart.git). 
Interrupt handler is this [UART1()](https://github.com/alexiionescu/esp32-wifi-uart/blob/bdd48eb4ef9fac46a3e72c53e3c72739c597c270/src/main.rs#L159) and executes [add_byte](https://github.com/alexiionescu/esp32-wifi-uart/blob/bdd48eb4ef9fac46a3e72c53e3c72739c597c270/src/uart_intr.rs#L28) for each byte.
I do nothing in the interrupt code but check some bytes and push most of them into a vector for later processing.
The same code, without running esp-wifi, has 0 errors.

Also this does not explain why on one restart I get 8% error rate (24h run), then another restart I get 67% (today starting, I will keep it for another 24h to see if the error rate drops)

manio, 2023-08-17:
Your interrupt handler looks fine for me...

alexiionescu, 2023-08-18:
I have updated my test project [esp32-wifi-uart](https://github.com/alexiionescu/esp32-wifi-uart.git)  with a very good scenario to reproduce.
I needed to implement a function in `esp-hal` to change the UART_RX_SIZE in the UART_MEM_CONF_REG from 1(default) to 2. 
I think this means increasing the UART1 RX FIFO RAM to 256 (?). It should be OK if I do not use UART2.

With this change, regardless if I use async mode or not, program gets stuck in esp-wifi at first uart1 interrupt call (my guess)
Rarely, however, it starts. And when that happens I get no uart1 data corruption. I have left such a rare successful run over the weekend in async mode.

Simple way to reproduce is use --features esp32,uart_intr,esp_now (interrupt version, which enable the interrupt immediately)
You will see that "esp_now init" is the last line to be printed. Then the program gets stuck.

If you use async version (--features esp32,esp32_async,uart,esp_now) the program gets stuck a little bit later when async mode enabled the interrupt at first `read` call.

Again, without `esp-wifi`, the uart works without issues.

So my suspicion is that somehow the esp-wifi and/or esp_now and UART RF FIFO RAM Memory have a collision of some sort. Unfortunately, I cannot explain this, cause I do not know the internal of the wi-fi tasks, not sure if they are documented anywhere public. In the technical reference manual I cannot find anything about wifi or esp_now.

And I hope the setting of UART_RX_SIZE = 0x02 is correct, the documentation is very poor: 
```
UART_RX_SIZE This register is used to configure the amount of memory allocated to the receive FIFO. The default number is 128 bytes. (R/W)
```
The UART_RX_SIZE has only 4 bits and the default value is 0x01 (I read it and it is correct).

manio, 2023-09-21:
@alexiionescu 
One quite important fix just get merged (https://github.com/esp-rs/esp-hal/pull/804).
Maybe it's worth to try it also in your case?

alexiionescu, 2023-09-21:
> @alexiionescu
> One quite important fix just get merged ([esp-rs/esp-hal#804](https://github.com/esp-rs/esp-hal/pull/804)).
> Maybe it's worth to try it also in your case?

Thanks, I was planning to run on latest versions when I get rid of my current tasks.
My problem happens only when running with `esp-wifi` enabled with `esp-hal` only works just fine. 
But who knows.

alexiionescu, 2023-11-08:
Working with latest `esp-hal` 0.13.1 and latest `esp-wifi`
I am closing this.

closed 2023-11-08 by alexiionescu