esp-rs/esp-wifi-sys #219: Bluetooth client api/example

ccruquan, 2023-07-27:
Is any plan support device as bluetooth client?

bjoernQ, 2023-08-07:
There are currently no real plans to add an example for working as a BLE central device.

The problem is for that we just don't have a convenient BLE stack which could do that. 
I think porting Nimble would be possible but ideally someone would create a pure Rust BLE central stack (but I don't know of anyone working on that right now)

I know that someone is definitely using esp-wifi to scan for BLE advertisements at least - if that is sufficient for you, we could add an example for that by just using raw HCI commands sent to the driver

trembel, 2023-12-21:
I think not only client as central, but also client as peripheral would be interesting (e.g. for Current Time Service, where the central, e.g a smartphone, is the server of the CTS and the device as a peripheral fetches the time information in the role of a client).

MabezDev, 2024-02-20:
I think having an example like this would be really nice too! If someone has some spare cycles, we'd appreciate some help in the form of contributions. We'd be happy to mentor and answer questions to help through the process, we just don't have time to work on this ourselves right now.