esp-rs/esp-wifi-sys #227: Is it possible to make ESP-NOW duplex

M4tsuri, 2023-08-07:
I wonder if it's possible to make ESP-NOW duplex, i.e. implement a `split` method that returns owned `Sender` and `Receiver`.

Here is a simple use case:

Suppose I have two tasks, one sending and one receiving:

```rust
async fn listener(radio: &'static Mutex<EspNow<'static>>) {
    let mut lock = radio.lock().await;
    _ = lock.receive().await;
}

async fn sender(radio: &'static Mutex<EspNow<'static>>) {
    let mut timer = interval(Duration::from_millis(1000));

    loop {
        timer.tick().await;
        radio.lock().await.send(&[]);
    }
}
```

When listener is waiting for new data, the lock is acquired by listener even when sender wants to send, which blocks sender from sending data. Currently, we need additional tweaks to make this work as excepted, for example, using channels.

However, I wonder if we could to this:

```rust
pub trait EspNowMode {}
pub trait WithRx: EspNowMode {}
pub trait WithTx: EspNowMode {}

pub struct RxOnly;

impl EspNowMode for RxOnly {}
impl WithRx for RxOnly {}

pub struct TxOnly;

impl EspNowMode for TxOnly {}
impl WithTx for TxOnly {}

pub struct TxRx;

impl EspNowMode for TxRx {}
impl WithRx for TxRx {}
impl WithTx for TxRx {}

pub struct NoTxRx;

impl EspNowMode for NoTxRx {}

pub struct EspNowTx<'d>(PhantomData<&'d ()>);
pub struct EspNowRx<'d>(PhantomData<&'d ()>);

pub struct EspNow<'d, M: EspNowMode> {
    _device: Option<PeripheralRef<'d, esp_hal_common::radio::Wifi>>,
    mode: core::marker::PhantomData<M>,
}

impl<'d> EspNow<'d, TxRx> {
    pub fn split(mut self) -> (EspNow<'d, NoTxRx>, EspNowTx<'d>, EspNowRx<'d>) {
        ESP_NOW_RC.fetch_add(2, Ordering::Relaxed);
        (
            EspNow { _device: self._device.take(), mode: PhantomData }, 
            EspNowTx(PhantomData), 
            EspNowRx(PhantomData)
        )
    }
}
```

And then implement `send` for `EspNowTx` and `receive(_async)` for `EspNowRx`. The returned `EspNow<'d, NoTxRx>` can still call the configuration methods. 

I tried to implement these in [my fork](https://github.com/M4tsuri/esp-wifi/blob/main/esp-wifi/src/esp_now/mod.rs), you can check it out.

Thank you.

M4tsuri, 2023-08-07:
I asked the developers of esp-now and they said that esp-now send is thread safe (https://github.com/espressif/esp-now/issues/87). So I think my idea is safe to implement.

Moreover, they mentioned that we **must** wait for the send callback of `esp_now_send`. I noticed that current implementation does not count this. Maybe there should be another issue for this.

bjoernQ, 2023-08-07:
Thanks! Generally, this sounds like a good idea to me

> Moreover, they mentioned that we must wait for the send callback of esp_now_send. I noticed that current implementation does not count this. Maybe there should be another issue for this.

Didn't knew that t.b.h. - I agree adding an issue for this is a good thing

M4tsuri, 2023-08-07:
Ok. I will open a pull request to the previous one after testing my implementation on the hardware. As for the latter one, I will write a demo to check the potential impact of a fix and open another issue later.

M4tsuri, 2023-08-11:
[esp-idf-svc](https://github.com/esp-rs/esp-idf-svc/blob/master/src/espnow.rs) implemented esp now in a way where all config, send and receive methods do not take mutable reference of `self`. Why not do what they did so we can just share references of `EspNow` struct between threads.

bjoernQ, 2023-08-11:
> [esp-idf-svc](https://github.com/esp-rs/esp-idf-svc/blob/master/src/espnow.rs) implemented esp now in a way where all config, send and receive methods do not take mutable reference of `self`. Why not do what they did so we can just share references of `EspNow` struct between threads.

To be honest I never looked into that but I think it's an easy and good enough solution for the problem. I don't see why we shouldn't do the same 👍

M4tsuri, 2023-08-11:
Closed in favor of #237

M4tsuri, 2023-08-11:
@bjoernQ  Sorry I just realized that I've made a fundamentally wrong design in #232, which makes `EspNow` totally unusable when performing data sending in multiple tasks (Sorry for not having it tested thoroughly).

Here I used an `AtomicWaker` to wake the future when callback is invoked.

https://github.com/esp-rs/esp-wifi/blob/e685dfcb08f249f5c6402b2d8ebb5a6a152f40bd/esp-wifi/src/esp_now/mod.rs#L660

However, what I didn't considered is the `register` method overwrites the previous waker. So when two tasks task1 and task2 are performing data sending, the following event series can occur:

```
[task1] esp_now_send
[task1] SendFuture::poll -> ESP_NOW_TX_WAKER = task1.waker
[task2] esp_now_send
[task2] SendFuture::poll -> ESP_NOW_TX_WAKER = task2.waker
[task1] callback -> wakes task2
[task2] Future returns with status of sending in task1 
```

Finally, task1 will never get waked while task2 gets wrong status.

I believe your idea of **wait before sending** can help solving this issue. I'm currently working on this. Or we can just expose the interface for registering callbacks since I found it's way more challenging than I expected.

bjoernQ, 2023-08-12:
Ah I see. Sorry I haven't noticed that before.

Maybe the easiest fix would be to combine what we have (awaitable result) and wait before sending. That way it's always just one "in-flight" send and the user can check the result when needed

M4tsuri, 2023-08-12:
I agree that this is a good enough solution for the sync version. However, after having tried several different approaches for the async version, I realized that it's way more difficult than I've expected.

#### My approach

Initially I used a simple global flag to indicate whether the previous sending is completed, i.e. the send callback is invoked. Let's call it `PREV_SEND_COMPLETE`, we may implement the future like this:

```rust
fn poll(...) -> ... {
    if !PREV_SEND_COMPLETE {
        return Poll::Pending
    }
    ...
    esp_now_send(...)
    ...
} 
```

However, when the future is firstly polled with `Poll::Pending` returned, it will probably **never** get polled again since the future will never be awoke by a waker.

So we should register a waker for current future before returning `Poll::Pending`, and the waker should be called when the previous sending completes. Finally a chain-like relationship will be built between all spawned future:

[image]

Note that we need to add future to this chain when at its first poll.

This idea still have may drawbacks:

1. It's pretty complex. I'm not sure if it's suitable to implement it in this crate.
2. It requires dynamic allocation since the length of that chain is unknown. (or static sized queue with additional checks)

I have a feeling that this idea is over designed. But I cannot find a simpler design that really works.

#### Another Idea
 
If `send_async` takes mutable borrowed receiver and we adopt to the type state design [in my first comment](https://github.com/esp-rs/esp-wifi/issues/227#issue-1838629551). User will need a mutex to send data in multiple tasks, thus making the futures naturally non-overlapping. I think this would be better than synchronizing the futures inside the API implementation and also provides more flexibility.

bjoernQ, 2023-08-12:
I agree, the async implementation shouldn't be over engineered.

Your suggestion sounds good to me

closed 2023-09-07 by M4tsuri