esp-rs/esp-wifi-sys #241: ESP32 disconnects right after connecting to wifi

sikora77, 2023-08-16:
After trying the example with:
`cargo espflash flash --example dhcp --release --features "embedded-svc,wifi" --monitor`

I get this output:

```
[2023-08-16T21:04:56Z INFO ] Serial port: '/dev/ttyUSB0'
[2023-08-16T21:04:56Z INFO ] Connecting...
[2023-08-16T21:04:56Z INFO ] Using flash stub
    Finished release [optimized + debuginfo] target(s) in 0.16s
Chip type:         esp32 (revision v1.0)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, Coding Scheme None
MAC address:       24:0a:c4:58:4d:90
App/part. size:    505,440/4,128,768 bytes, 12.24%
[00:00:01] [========================================]      17/17      0x1000                         [00:00:00] [========================================]       1/1       0x8000                         [00:00:30] [========================================]     320/320     0x10000                        [2023-08-16T21:05:30Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7104
0x3fff0030 - _ZN8esp_wifi9HEAP_DATA17hb30a9480f2f25a13E
    at ??:??
load:0x40078000,len:15576
0x40078000 - __bswapsi2
    at ??:??
load:0x40080400,len:4
0x40080400 - _init_end
    at ??:??
ho 8 tail 4 room 4
load:0x40080404,len:3876
0x40080404 - _init_end
    at ??:??
entry 0x4008064c
0x4008064c - phy_enter_critical
    at /home/sikora/.cargo/git/checkouts/esp-wifi-836f3b2af57fa847/6b53683/esp-wifi/src/common_adapter/common_adapter_esp32.rs:181
I (30) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (31) boot: compile time Jun  7 2023 07:48:23
I (33) boot: Multicore bootloader
I (37) boot: chip revision: v1.0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 003f0000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=13430h ( 78896) map
I (134) esp_image: segment 1: paddr=00023458 vaddr=3ffb0000 size=01084h (  4228) load
I (136) esp_image: segment 2: paddr=000244e4 vaddr=3ffbbd74 size=00190h (   400) load
I (140) esp_image: segment 3: paddr=0002467c vaddr=40080000 size=0ad10h ( 44304) load
I (167) esp_image: segment 4: paddr=0002f394 vaddr=00000000 size=00c84h (  3204) 
I (168) esp_image: segment 5: paddr=00030020 vaddr=400d0020 size=5b618h (374296) map
I (313) boot: Loaded app from partition at offset 0x10000
I (313) boot: Disabling RNG early entropy source...
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100 }
wifi_set_configuration returned Ok(())
is wifi started: Ok(true)
Start Wifi Scan
AccessPointInfo { ssid: "SSID1", bssid: [72, 34, 84, 23, 152, 34], channel: 11, secondary_channel: None, signal_strength: -51, protocols: EnumSet(), auth_method: WPA2Personal }
AccessPointInfo { ssid: "SSID2", bssid: [84, 219, 162, 38, 179, 161], channel: 11, secondary_channel: None, signal_strength: -75, protocols: EnumSet(), auth_method: WPA2Personal }
AccessPointInfo { ssid: "SSID3", bssid: [200, 234, 248, 13, 135, 30], channel: 5, secondary_channel: Above, signal_strength: -77, protocols: EnumSet(), auth_method: WPA2Personal }
Ok(EnumSet(Client | AccessPoint))
wifi_connect Ok(())
Wait to get connected
Disconnected

```

The password is definitely set correctly and the esp disconnects almost instantly

bjoernQ, 2023-08-17:
Which toolchain version are you using? (`cargo +esp --version`)

The crash looks like some miscompilation / misoptimization but I haven't seen them for a while on ESP32.

The unsuccessful connect: So, it sometimes succeeds and sometimes not, right? It can happen for multiple reasons and esp-wifi doesn't automatically reconnects / retries. It's left to the user's code but the examples don't do it (for simplicity).

Could you try running with the `wifi-logs` feature activated to see what the WiFi driver is actually doing?

sikora77, 2023-08-17:
Nevermind, the password was just exported the wrong way, everything works now, sorry to bother you

closed 2023-08-17 by bjoernQ