esp-rs/esp-wifi-sys #242: Wifi randomly crashing

sikora77, 2023-08-19:
After running for some time i get the following panic
```
!! A panic occured in '/home/sikora/.cargo/git/checkouts/esp-wifi-836f3b2af57fa847/6b53683/esp-wifi/src/wifi/mod.rs', at line 119, column 9
 
PanicInfo {
    payload: Any { .. },
    message: Some(
        range end index 1514 out of range for slice of length 1510,
    ),
    location: Location {
        file: "/home/sikora/.cargo/git/checkouts/esp-wifi-836f3b2af57fa847/6b53683/esp-wifi/src/wifi/mod.rs",
        line: 119,
        col: 9,
    },
    can_unwind: true,
}
 
Backtrace:
```

It seems completely random and unrelated to the code that I am running

bjoernQ, 2023-08-20:
Ok we should probably check the size before copying. But it's a bit strange anyway.
You could increase the MTU (see https://github.com/esp-rs/esp-wifi/blob/main/docs/tuning.md) to 1500 or 1514

sikora77, 2023-08-26:
changing the mtu to 1500 seems to have fixed it, setting it any higher makes the device unable to obtain an ip address

closed 2024-02-20 by MabezDev