esp-rs/esp-wifi-sys #246: Update for embassy-next

28Smiles, 2023-08-22:
Embassy imtroduced its own `Driver` trait, simmilar to the smoltcp trait. See: https://github.com/embassy-rs/embassy/blob/main/embassy-net-driver/src/lib.rs#L26

This should be straight forward to implement, but would require using the current version from the github repository instead of from crates.io . Any Suggestions?

MabezDev, 2023-08-22:
It's already implemented here: https://github.com/esp-rs/esp-wifi/blob/b8bddffa46605a082610e8fc2fea90af98dacb3d/esp-wifi/src/wifi/mod.rs#L1323C25-L1323C25

closed 2023-08-22 by 28Smiles

28Smiles, 2023-08-22:
I already activated this feature but the trait is not found, and i dont get why, the bound should be fullfilled, maybe you can help me.

```
error[E0277]: the trait bound `WifiDevice<'static>: embassy_net::embassy_net_driver::Driver` is not satisfied
   --> src\main.rs:177:26
    |
177 | async fn net_task(stack: &'static Stack<WifiDevice<'static>>) {
    |                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `embassy_net::embassy_net_driver::Driver` is not implemented for `WifiDevic
e<'static>`
    |
    = help: the trait `embassy_net::embassy_net_driver::Driver` is implemented for `&mut T`
note: required by a bound in `Stack`
   --> C:\Users\Leon\.cargo\git\checkouts\embassy-9312dcb0ed774b29\b3212ae\embassy-net\src\lib.rs:206:21
    |
206 | pub struct Stack<D: Driver> {
    |                     ^^^^^^ required by this bound in `Stack`
```

bjoernQ, 2023-08-23:
Oh, the embassy-next branch is quite old .... I should delete it.

You might try `cargo update` and `cargo clean`