esp-rs/esp-wifi-sys #247: Document the required rustflags

laundmo, 2023-08-23:
it seems the following rustflags is required to successfully compile this repo:

```toml
rustflags = [
  # ...
  "-C", "link-arg=-Tesp32s3_rom_functions.x",
  #                      ^^ change this
]
```
After searching this repo for a while i found
- #99

which explains that this is neccessary

bjoernQ, 2023-08-24:
We have this information in the README.md here: https://github.com/esp-rs/esp-wifi#link-configuration

Closing this - feel free to re-open if needed

closed 2023-08-24 by bjoernQ

laundmo, 2023-08-24:
from that it's not clear what format it should have:
esp32_s3_rom_functions

bjoernQ, 2023-08-24:
it's just `Trom_functions.x` .... previously we had the chip's name in it but we changed that. 

So, you might find examples in other repositories which are using an older version. 

Don't know how to make the README clearer in that regard but if you have an idea how to rephrase it to make it clearer, we are happy to get an PR