esp-rs/esp-wifi-sys #250: ESP32C3 dhcp example hangs at `Wait to get connected`

n8henrie, 2023-08-25:
```console
$ git rev-parse HEAD
b8bddffa46605a082610e8fc2fea90af98dacb3d
$ cd examples-esp32c3
$ cargo espflash flash --monitor --example dhcp --release --features "embedded-svc,wifi"
[2023-08-25T20:40:42Z INFO ] Serial port: '/dev/tty.usbserial-1110'
[2023-08-25T20:40:42Z INFO ] Connecting...
[2023-08-25T20:40:43Z INFO ] Using flash stub
    Finished release [optimized + debuginfo] target(s) in 0.17s
Chip type:         esp32c3 (revision v0.3)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       84:f7:03:39:f1:cc
App/part. size:    425,920/4,128,768 bytes, 10.32%
[00:00:01] [========================================]      13/13      0x0
[00:00:00] [========================================]       1/1       0x8000
[00:00:28] [========================================]     251/251     0x10000                                                                                                                                                                 [2023-08-25T20:41:14Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5820,len:0x171c
0x3fcd5820 - _stack_start
    at ??:??
load:0x403cc710,len:0x968
0x403cc710 - .L17
    at ??:??
load:0x403ce710,len:0x2f68
0x403ce710 - .L17
    at ??:??
SHA-256 comparison failed:
Calculated: 1d06b938c0222bf626e0bdf46178b1b37ab24d03f0360fc8fcf7153c2571deaf
Expected: 68d7bdf643ba446b8ed7ae8423241d442fd052b2bc77091100ba06fd65dcf8d5
Attempting to boot anyway...
entry 0x403cc710
0x403cc710 - .L17
    at ??:??
I (50) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (50) boot: compile time Jun  7 2023 07:59:10
I (51) boot: chip revision: v0.3
I (55) boot.esp32c3: SPI Speed      : 40MHz
I (60) boot.esp32c3: SPI Mode       : DIO
I (65) boot.esp32c3: SPI Flash Size : 4MB
I (69) boot: Enabling RNG early entropy source...
I (75) boot: Partition Table:
I (78) boot: ## Label            Usage          Type ST Offset   Length
I (86) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (93) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (101) boot:  2 factory          factory app      00 00 00010000 003f0000
I (108) boot: End of partition table
I (112) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=0e1e4h ( 57828) map
I (134) esp_image: segment 1: paddr=0001e20c vaddr=3fc84488 size=0120ch (  4620) load
I (135) esp_image: segment 2: paddr=0001f420 vaddr=3fca0da0 size=00168h (   360) load
I (139) esp_image: segment 3: paddr=0001f590 vaddr=40380000 size=00a88h (  2696) load
I (149) esp_image: segment 4: paddr=00020020 vaddr=42000020 size=5456ch (345452) map
I (232) esp_image: segment 5: paddr=00074594 vaddr=40380a88 size=039fch ( 14844) load
I (237) boot: Loaded app from partition at offset 0x10000
I (237) boot: Disabling RNG early entropy source...
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100 }
wifi_set_configuration returned Ok(())
is wifi started: Ok(true)
Start Wifi Scan
AccessPointInfo { ssid: REDACTED, bssid: REDACTED, channel: 6, secondary_channel: None, signal_strength: -34, protocols: EnumSet(), auth_method: WPAWPA2Personal }
Ok(EnumSet(Client | AccessPoint))
wifi_connect Ok(())
Wait to get connected
```

It hangs indefinitely here. SSID and PASSWORD are correct, triple checked, and the access point is 10 feet away and provides a 2.4 GHz network.

I've previously tested this project on the same network and was able to get connected, but it's been a while since I tinkered with it!

n8henrie, 2023-08-25:
I've occasionally gotten this as well:

```
ERROR - RX QUEUE FULL
ERROR - RX QUEUE FULL
ERROR - RX QUEUE FULL
ERROR - RX QUEUE FULL
ERROR - RX QUEUE FULL
```

Running with the `wifi-logs` feature, it doesn't *look* like it's connecting, I see a lot of:

```
INFO - V (1455) wifi:
INFO - scan any ssid
INFO -

INFO - V (1462) wifi:
INFO -

INFO - start min timer
INFO -

INFO - V (1470) wifi:
INFO - start max timer
INFO -

INFO - V (1484) wifi:
INFO - probe ap already in q: 60:b7:6e:20:51:a3, discard
INFO -

INFO - D (1497) wifi:
INFO - scan end: arg=0x0, status=0, ss_state=0x03
INFO -

INFO - D (1498) wifi:
INFO - perform scan: ss_state=0x09, chan<12,0>, dur<10,20>
INFO -

INFO - V (1507) wifi:
INFO - change: chan<12,0>, dur<10,20>
INFO -

INFO - V (1515) wifi:
INFO - ht20 freq=2467, chan=12
INFO -

INFO - V (1524) wifi:
INFO - enter start op, arg=0x3fc9f224
0x3fc9f224 - .LANCHOR45
    at ??:??
INFO -

INFO - V (1531) wifi:
INFO - scan operation: state=0x03, chan<12,0>, arg=0x3fc9f224, status=0
0x3fc9f224 - .LANCHOR45
    at ??:??
INFO -

INFO - V (1542) wifi:
INFO - scan any ssid
INFO -
```

bjoernQ, 2023-08-27:
Seems like it doesn't see the SSID it is looking for. Is the access point found in the output of the WiFi scan?

If yes - is the SSID correctly set as an env-variable? e.g. if it contains special characters like `$` your shell might try to do variable substitution

The `ERROR - RX QUEUE FULL` shouldn't get logged at error level. It's not really an error and we should probably change it to warning level - you could increase the rx-queue size but I don't think that's the problem here

n8henrie, 2023-08-28:
Hi -- thanks for your time!

> Seems like it doesn't see the SSID it is looking for. Is the access point found in the output of the WiFi scan?

Sometimes, not always, no difference in output or behavior whether or not the SSID is in the output. The signal strength is about -40. 

> If yes - is the SSID correctly set as an env-variable? e.g. if it contains special characters like $ your shell might try to do variable substitution

Yes, it's correct -- quadruple verified. Only "special" char in the SSID is an underscore. Even tried with a different network (guest network, same access point) with a totally different SSID and password, same result. 

> you could increase the rx-queue size but I don't think that's the problem here

I agree, I've seen this message two or so times out of several dozen attempts.

bjoernQ, 2023-08-29:
Very weird. Would be interesting if esp-idf (e.g. https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/station) examples connect successfully.

Otherwise, you could try to use your smartphone as an access-point (just for a test)

n8henrie, 2023-08-29:
>  Would be interesting if esp-idf (e.g. https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/station) examples connect successfully.

That example seems to work fine, connects and gets an IP address:

```console
$ idf.py -p /dev/tty.usbserial-110 monitor
Executing action: monitor
Running idf_monitor in directory /Users/n8henrie/git/esp-idf/examples/wifi/getting_started/station
Executing "/Users/n8henrie/.espressif/python_env/idf5.1_py3.11_env/bin/python /Users/n8henrie/git/esp-idf/tools/idf_monitor.py -p /dev/tty.usbserial-110 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c3 --revision 3 --decode-panic backtrace /Users/n8henrie/git/esp-idf/examples/wifi/getting_started/station/build/wifi_station.elf -m '/Users/n8henrie/.espressif/python_env/idf5.1_py3.11_env/bin/python' '/Users/n8henrie/git/esp-idf/tools/idf.py' '-p' '/dev/tty.usbserial-110'"...
--- WARNING: Serial ports accessed as /dev/tty.* will hang gdb if launched.
--- Using /dev/cu.usbserial-110 instead...
--- idf_monitor on /dev/cu.usbserial-110 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-202�ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1704
load:0x403cc710,len:0x968
load:0x403ce710,len:0x2f68
entry 0x403cc710
I (30) boot: ESP-IDF v5.1.1 2nd stage bootloader
I (30) boot: compile time Aug 29 2023 08:11:42
I (30) boot: chip revision: v0.3
I (33) boot.esp32c3: SPI Speed      : 80MHz
I (38) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 2MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (86) boot: End of partition table
I (90) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=1fb80h (129920) map
I (119) esp_image: segment 1: paddr=0002fba8 vaddr=3fc90600 size=00470h (  1136) load
I (120) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=8a1e0h (565728) map
I (215) esp_image: segment 3: paddr=000ba208 vaddr=3fc90a70 size=023c4h (  9156) load
I (217) esp_image: segment 4: paddr=000bc5d4 vaddr=40380000 size=105a4h ( 66980) load
I (239) boot: Loaded app from partition at offset 0x10000
I (239) boot: Disabling RNG early entropy source...
I (250) cpu_start: Unicore app
I (251) cpu_start: Pro cpu up.
I (259) cpu_start: Pro cpu start user code
I (259) cpu_start: cpu freq: 160000000 Hz
I (259) cpu_start: Application information:
I (262) cpu_start: Project name:     wifi_station
I (268) cpu_start: App version:      v5.1.1
I (273) cpu_start: Compile time:     Aug 29 2023 08:11:33
I (279) cpu_start: ELF file SHA256:  c3557f865735c9b9...
I (285) cpu_start: ESP-IDF:          v5.1.1
I (290) cpu_start: Min chip rev:     v0.3
I (294) cpu_start: Max chip rev:     v0.99
I (299) cpu_start: Chip rev:         v0.3
I (304) heap_init: Initializing. RAM available for dynamic allocation:
I (311) heap_init: At 3FC96EE0 len 00045830 (278 KiB): DRAM
I (317) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (324) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (331) spi_flash: detected chip: generic
I (335) spi_flash: flash io: dio
W (339) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (352) sleep: Configure to isolate all GPIO pins in sleep state
I (359) sleep: Enable automatic switching of GPIO sleep configuration
I (366) app_start: Starting scheduler on CPU0
I (371) main_task: Started on CPU0
I (371) main_task: Calling app_main()
I (381) wifi station: ESP_WIFI_MODE_STA
I (381) pp: pp rom version: 9387209
I (381) net80211: net80211 rom version: 9387209
[... 25 lines cut ...]
I (531) wifi station: wifi_init_sta finished.
I (571) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (571) wifi:state: init -> auth (b0)
I (571) wifi:state: auth -> init (8a0)
I (571) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (581) wifi station: retry to connect to the AP
I (581) wifi station: connect to the AP fail
I (2991) wifi station: retry to connect to the AP
I (2991) wifi station: connect to the AP fail
I (3021) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (3031) wifi:state: init -> auth (b0)
I (3031) wifi:state: auth -> assoc (0)
I (3051) wifi:state: assoc -> run (10)
I (3081) wifi:connected with REDACTED, aid = 7, channel 6, BW20, bssid = e4:d4:2a:b5:a8:3a
I (3091) wifi:security: WPA2-PSK, phy: bgn, rssi: -45
I (3091) wifi:pm start, type: 1

I (3091) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
I (3101) wifi:<ba-add>idx:0 (ifx:0, e4:c3:2a:e5:a8:3a), tid:0, ssn:0, winSize:64
I (3131) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (4101) esp_netif_handlers: sta ip: 192.168.0.218, mask: 255.255.255.0, gw: 192.168.0.2
I (4101) wifi station: got ip:192.168.0.218
I (4101) wifi station: connected to ap SSID:REDACTED password:REDACTED
I (4111) main_task: Returned from app_main()
```

bjoernQ, 2023-08-29:
But interesting it fails a few times before it succeeds. I'd say increasing the reconnect tries might help but ... in the wifi-logs above it doesn't even try to connect and keeps scanning 🤔 or was that from the beginning of the log? The example contains a general scan which might be what we see there

n8henrie, 2023-08-31:
Just checked my router logs, I don't see any DHCP requests from the esp32's MAC address.

bjoernQ, 2023-08-31:
> Just checked my router logs, I don't see any DHCP requests from the esp32's MAC address.

Yes, in the wifi-log at the top it doesn't seem it even gets to send an associate request to the router

I'm really running out of ideas. Maybe one thing: You could try to see what happens if you explicitly set the channel in `client_config` (i.e. `channel: Some(6u8)`). 

Another thing (given even esp-idf struggles a little bit to connect): can you try to move the ESP32-C3 closer to the AP?

n8henrie, 2023-08-31:
Couldn't get it to work from my phone's wifi hotspot either.

After a hundred or so tries it suddenly started working (intermittently at least) with my main access point with no code changes.

> can you try to move the ESP32-C3 closer to the AP?

Not really, I'm about 10 feet away with direct line of site. Actually I seem to remember a similar problem with an esp8266 at some point that resolved by moving *farther* away from the AP, so I went and tinkered a room away and that's when it started working. No idea if that was related or just coincidental.

Thanks for your time and attention, sorry to not have anything more helpful to report!

closed 2023-08-31 by n8henrie