esp-rs/esp-wifi-sys #256: Large static objects occupying dram

M4tsuri, 2023-09-02:
I noticed that static objects in this crate takes about 70% of `dram_seg` segment on my ESP32-S2-WROOM-I, which could cause unexpected overflow when user heavily uses static variables:

[image]

More precisely, the first three are defined in 

1. https://github.com/esp-rs/esp-wifi/blob/fd172502cdef96ab9e464cc5c9b072106d8dc00d/esp-wifi/src/lib.rs#L159
2. https://github.com/esp-rs/esp-wifi/blob/fd172502cdef96ab9e464cc5c9b072106d8dc00d/esp-wifi/src/wifi/mod.rs#L105
3. https://github.com/esp-rs/esp-wifi/blob/fd172502cdef96ab9e464cc5c9b072106d8dc00d/esp-wifi/src/preempt/mod.rs#L20

I come up with two ideas on this situation:

1. Documentation of dram usage is necessary. I got an overflow error in linking stage and spent plenty of time searching for what takes so much memory and then solved it by reducing `rx_queue_size`, which in turn reduces size of `DATA_FRAME_BACKING_MEMORY`.
2. We can put these objects somewhere else. For example, there is a unused hole in dram of esp32s2 starting at `0x3ffeab00` with a length of `0x15500`. Maybe this is a good place for the large objects.

About the unused hole:

The end of bootloader memory is at `0x3ffeab00`, which is defined [here](https://github.com/espressif/esp-idf/blob/82cceabc6e6a0a2d8c40e2249bc59917cc5e577a/components/bootloader/subproject/main/ld/esp32s2/bootloader.ld#L15). It seems nothing is placed behind the bootloader so there is a space with length of `0x40000000 - 0x3ffeab00 = 0x15500`. I tried to place some static objects here and it works.

bjoernQ, 2023-09-04:
Using the currently unused memory on ESP32-S2 definitely makes sense. We have a different but similar situation on ESP32 where we put the wifi-heap into the unused memory area

MabezDev, 2023-09-04:
@M4tsuri I have some ideas to drastically reduce the memory usage (hopefully by about half if my experiments are successful). Out of interest, which tool are you using to get that nice output of each symbol's usage? I know it's data from `nm`, but that looks super nice.

MabezDev, 2023-09-04:
@M4tsuri Feel free to try out this PR: https://github.com/esp-rs/esp-wifi/pull/259

M4tsuri, 2023-09-05:
> @M4tsuri I have some ideas to drastically reduce the memory usage (hopefully by about half if my experiments are successful). Out of interest, which tool are you using to get that nice output of each symbol's usage? I know it's data from `nm`, but that looks super nice.

I generated a csv from readelf output with a command like this:

```
echo "Num,Value,Size,Type,Bind,Vis,Ndx,Name" > output.csv && ls ../target/xtensa-esp32s2-none-elf/
release/deps/*.rlib | xargs -I {} readelf --sym-base=10 -s {} -W | grep OBJECT | sed -E "s/ +/, /g" 
| sed -E "s/^, //g" >> output.csv
```

M4tsuri, 2023-09-05:
> @M4tsuri Feel free to try out this PR: #259

Thank you. This looks promising. I'll try it soon.

M4tsuri, 2023-09-06:
As #259 get merged, the second large object is gone. I'll try to move `HEAP_DATA` and `TASK_STACK` to unused memory areas. 

[image]

MabezDev, 2024-02-20:
I think this is mostly resolved, but further improvements are welcome :)

closed 2024-02-20 by MabezDev