esp-rs/esp-wifi-sys #270: Compile failure esp32s3 esp-now e.g.: invalid register name

Ben-PH, 2023-09-22:
```
git clone https://github.com/esp-rs/esp-wifi.git
cd esp-wifi/examples-esp32s3
cargo build --example esp_now --release --features "esp-now"
```

error:

```
❯ cargo build --example esp_now --release --features "esp-now"
   Compiling esp-wifi v0.1.0 (/home/ben/rust/esp-stuff/esp-wifi/esp-wifi)
   Compiling examples-esp32s3 v0.1.0 (/home/ben/rust/esp-stuff/esp-wifi/examples-esp32s3)
error: invalid register name
   --> esp-wifi/src/timer_esp32s3.rs:196:27
    |
196 |         core::arch::asm!("wsr.intset  {0}", in(reg) intr, options(nostack));
    |                           ^
    |
note: instantiated into assembly here
   --> <inline asm>:1:2
    |
1   |     wsr.intset  a8
    |     ^

error: could not compile `esp-wifi` (lib) due to previous error
```

then:

`git checkout 87687c82db75bb7378957c48d990602f423c7d36`

build works. checkout the next commit that edits the file:

`git checkout 5d0c416c0d6937acf8813306a419fb270c830715`
...and the error re-appears.

seems to be this diff:

https://github.com/esp-rs/esp-wifi/commit/5d0c416c0d6937acf8813306a419fb270c830715#diff-526af10fb97077f069ba729e126156e3b4a38b7650f98c5c92835d2e063bb456L196-R196

MabezDev, 2023-09-22:
You need to update your esp compiler version, `inset` was added in 1.70 IIRC.

closed 2023-09-22 by MabezDev

Ben-PH, 2023-09-22:
> You need to update your esp compiler version, `inset` was added in 1.70 IIRC.

...I should have checked out the PR comments: https://github.com/esp-rs/esp-wifi/pull/248#issuecomment-1700464301

I'm on v1.70, it's v1.72 I need