esp-rs/esp-wifi-sys #274: assistance with "esp32" example "embassy_dhcp.rs"

Token-Thinker, 2023-10-01:
Hello, 

Trying to get the example for the "embassy_dhcp" up and running for the esp32, however running into a slight snag that I can't figure out for the life of me. Any hint or advice would be truly appreciated. 

Code Snippet Area:
```
#[embassy_executor::task]
async fn net_task(stack: &'static Stack<WifiDevice<'static>>) {
    stack.run().await
}

#[embassy_executor::task]
async fn task(stack: &'static Stack<WifiDevice<'static>>) {
    let mut rx_buffer = [0; 4096];
    let mut tx_buffer = [0; 4096];
```

Error:
```
the trait bound `WifiDevice<'static>: embassy_net::embassy_net_driver::Driver` is not satisfied
the trait `embassy_net::embassy_net_driver::Driver` is implemented for `&mut T`rustc[Click for full compiler diagnostic](rust-analyzer-diagnostics-view:/diagnostic%20message%20%5B4%5D?4#file%3A%2F%2F%2Fhome%2Fmguerrier%2FDocuments%2Fworkspace-rust%2Ftkr-app%2Fsrc%2Fmain.rs)
lib.rs(200, 21): required by a bound in `embassy_net::Stack`
```

manifest:
```
[profile.dev.package.esp-wifi]
opt-level = 3

[profile.dev]
lto = "off"
[profile.release]
lto = "off"

[dependencies]
hal = { package = "esp32-hal", version = "0.15.0", features = ["embassy","embassy-time-timg0","async"]}
esp-backtrace = { version = "0.8.0", features = ["esp32", "panic-handler", "exception-handler", "print-uart"] }
esp-println = { version = "0.6.0", features = ["esp32","log"] }
log = { version = "0.4.18" }
esp-alloc = { version = "0.3.0" }
esp-wifi  = { git = "https://github.com/esp-rs/esp-wifi/", rev = "fbb8417", features = ["esp32","wifi-logs", "wifi"] }
smoltcp = { version = "0.10.0", default-features=false, features = ["proto-igmp", "proto-ipv4", "socket-tcp", "socket-icmp", "socket-udp", "medium-ethernet", "proto-dhcpv4", "socket-raw", "socket-dhcpv4"] }
embedded-svc = { version = "0.25.0", default-features = false, features = [] }
embedded-io = "0.4.0"
heapless = { version = "0.7.14", default-features = false }

embassy-executor = { version = "0.3.0", package = "embassy-executor", features = ["nightly", "executor-thread", "integrated-timers"] }
embassy-time = { version = "0.1.3", features = ["nightly"] }
embassy-net = { version = "0.1.0", features = ["nightly", "tcp", "udp", "dhcpv4", "medium-ethernet"] }[profile.dev.package.esp-wifi]
opt-level = 3

[profile.dev]
lto = "off"
[profile.release]
lto = "off"

[dependencies]
hal = { package = "esp32-hal", version = "0.15.0", features = ["embassy","embassy-time-timg0","async"]}
esp-backtrace = { version = "0.8.0", features = ["esp32", "panic-handler", "exception-handler", "print-uart"] }
esp-println = { version = "0.6.0", features = ["esp32","log"] }
log = { version = "0.4.18" }
esp-alloc = { version = "0.3.0" }
esp-wifi  = { git = "https://github.com/esp-rs/esp-wifi/", rev = "fbb8417", features = ["esp32","wifi-logs", "wifi"] }
smoltcp = { version = "0.10.0", default-features=false, features = ["proto-igmp", "proto-ipv4", "socket-tcp", "socket-icmp", "socket-udp", "medium-ethernet", "proto-dhcpv4", "socket-raw", "socket-dhcpv4"] }
embedded-svc = { version = "0.25.0", default-features = false, features = [] }
embedded-io = "0.4.0"
heapless = { version = "0.7.14", default-features = false }

embassy-executor = { version = "0.3.0", package = "embassy-executor", features = ["nightly", "executor-thread", "integrated-timers"] }
embassy-time = { version = "0.1.3", features = ["nightly"] }
embassy-net = { version = "0.1.0", features = ["nightly", "tcp", "udp", "dhcpv4", "medium-ethernet"] }
```

MabezDev, 2023-10-01:
The embassy-net version used in your project must match the version in esp-wifi.

Token-Thinker, 2023-10-01:
> The embassy-net version used in your project must match the version in esp-wifi.

Confused? That is the version out of the manifest in the repo itself, is there a different version that I am missing?

i got it from here -> https://github.com/esp-rs/esp-wifi/blob/main/Cargo.toml

bjoernQ, 2023-10-02:
Seems a few needed features weren't selected for `embassy-executor` and `esp-wifi`. This should work

```toml
[profile.dev.package.esp-wifi]
opt-level = 3

[profile.dev]
lto = "off"
[profile.release]
lto = "off"

[dependencies]
hal = { package = "esp32-hal", version = "0.15.0", features = ["embassy","embassy-time-timg0","async"]}
esp-backtrace = { version = "0.8.0", features = ["esp32", "panic-handler", "exception-handler", "print-uart"] }
esp-println = { version = "0.6.0", features = ["esp32","log"] }
log = { version = "0.4.18" }
esp-alloc = { version = "0.3.0" }
esp-wifi  = { git = "https://github.com/esp-rs/esp-wifi/", rev = "fbb8417", features = ["esp32", "wifi-logs", "wifi", "async", "embassy-net"] }
smoltcp = { version = "0.10.0", default-features=false, features = ["proto-igmp", "proto-ipv4", "socket-tcp", "socket-icmp", "socket-udp", "medium-ethernet", "proto-dhcpv4", "socket-raw", "socket-dhcpv4"] }
embedded-svc = { version = "0.25.0", default-features = false, features = [] }
embedded-io = "0.4.0"
heapless = { version = "0.7.14", default-features = false }

embassy-executor = { version = "0.3.0", package = "embassy-executor", features = ["nightly", "executor-thread", "integrated-timers", "arch-xtensa"] }
embassy-time = { version = "0.1.3", features = ["nightly"] }
embassy-net = { version = "0.1.0", features = ["nightly", "tcp", "udp", "dhcpv4", "medium-ethernet"] }
```

Token-Thinker, 2023-10-02:
Thank you!, I finally got new errors and going to take a crack at solving them.

closed 2023-10-02 by Token-Thinker