esp-rs/esp-wifi-sys #285: esp32-s3 overflow panic

Dav1dde, 2023-10-14:
Just letting the wifi idle for a while I got this panic using the latest master:

```
PanicInfo {
40000000
    payload: Any { .. },
    message: Some(
        attempt to multiply with overflow,
    ),
    location: Location {
        file: "/home/crypt-dav1d/dav1d/.cargo/git/checkouts/esp-wifi-836f3b2af57fa847/887eb22/esp-wifi/src/timer/xtensa.rs",
        line: 44,
        col: 5,
    },
    can_unwind: true,
}
 
Backtrace:
 
0x42072647
0x42072647 - core::panicking::panic
    at /home/crypt-dav1d/dav1d/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/panicking.rs:117
0x42024c1f
0x42024c1f - esp_wifi::timer::arch_specific::get_systimer_count
    at /home/crypt-dav1d/dav1d/.cargo/git/checkouts/esp-wifi-836f3b2af57fa847/887eb22/esp-wifi/src/timer/xtensa.rs:44
0x4202168e
0x4202168e - esp_timer_get_time
    at /home/crypt-dav1d/dav1d/.cargo/git/checkouts/esp-wifi-836f3b2af57fa847/887eb22/esp-wifi/src/wifi/os_adapter.rs:1183
0x4037b262
0x4037b262 - pm_tbtt_process
    at ??:??
0x4003d024
0x4003d024 - rom_rx_gain_force
    at ??:??
0x4037a4c0
0x4037a4c0 - ppTask
    at ??:??
0x4202465f
0x4202465f - core::sync::atomic::atomic_load
    at /home/crypt-dav1d/dav1d/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/sync/atomic.rs:3184
0x420246b4
0x420246b4 - esp_wifi::tasks::worker_task1
    at /home/crypt-dav1d/dav1d/.cargo/git/checkouts/esp-wifi-836f3b2af57fa847/887eb22/esp-wifi/src/tasks.rs:23
0x40000000
0x40000000 - ets_rom_layout_p
    at ??:??
```

Relevant function:
https://github.com/esp-rs/esp-wifi/blob/887eb224aaa8224df77f8ad4125a31adce62e8e0/esp-wifi/src/timer/xtensa.rs#L30-L45

bjoernQ, 2023-10-14:
cc @bugadani

bugadani, 2023-10-14:
huh... fun, thanks for the ping! @Dav1dde how much time does this need to manifest?

Dav1dde, 2023-10-14:
Unfortunately I only came back to the error, that was 1h and 7m after I flashed and rebooted, so the timeframe was within 1h and 7minutes, I'll see if I can get a better time for you.

bugadani, 2023-10-14:
No need, thank you, I'm running my own test now. Unless my saturday maths is off, it should be about half an hour.

bugadani, 2023-10-14:
Interestingly, I wasn't able to reproduce the panic, but my device inexplicably slowed down after some time. I can explain how the old code didn't panic and why my changes did, and I'm confident that the linked PR is a fix, but it would have been better to see the actual crash.

PR #287 (2023-10-18): Fix overflow in timestamp calculation
This should fix #285 - there's no more multiplication, so there should be no more overflow, either. Still testing the before state though just to make sure.

The original time keeping didn't panic for the same multiplication, because:
 - The global incrementing compiled down to an addition with a constant (2AAA_AAAA I believe)
 - The cycle counter is 32 bits casted to u64, multiplying that with 40_000_000 results in a 64 bit number without overflow, which was then divided by 240_000_000 without problem
 - The results were then the sum of these two values

 [... cut]
closed 2023-10-18 by MabezDev