esp-rs/esp-wifi-sys #290: Option to disable unneeded socket types

bugadani, 2023-10-16:
Most of smoltcp's features should probably be either opt-in, or opt-out: UDP, TCP, DNS, DHCP, IGMP are all technically optional. We should give users the option to trim the library down a bit if they want to.

bjoernQ, 2023-10-17:
I agree - we already have an opt-in for Ipv6 but we should expand that to the same granularity as embassy-net

Ideally, we can set the defaults in a way that users don't need to worry too much

BryanKadzban, 2023-10-30:
I was actually just coming here to propose something that this would also work for -- in my case I only need esp-wifi to support sending raw wifi frames, no upper layer protocols at all.  (I have an existing stack that I'm trying to use with one of these devices as the MAC/PHY layer.)  If I can disable everything, it would help a lot.

(I currently have a fork of this repo where I've simply deleted all the smoltcp-using code, but that's been very hard to maintain.  Changing it all to be feature-gated would make it easier to carry the patches (rebases would be far simpler to resolve), and would probably be a group of commits that could be merged here, actually.  Not sure when I'll get it ready, but I'll start working on it soon.)

PR #340 (2023-11-13): Add features to disable various network protocols
smoltcp has feature flags to allow enabling and disabling several protocols.  Expose those feature flags to users of esp-wifi as well, to let them trim down both esp-wifi and smoltcp.

Also added a loop to the CI config to build with all combinations of all these new feature flags, to ensure all the possible combinations continue to work.

Fixes #290
closed 2023-11-13 by bjoernQ