esp-rs/esp-wifi-sys #291: ERROR - RX QUEUE FULL

galah92, 2023-10-17:
Consider the following code:
```rust
#![no_std]
#![no_main]

use embedded_svc::wifi::{ClientConfiguration, Configuration, Wifi};
use esp_backtrace as _;
use esp_println::println;
use esp_wifi::wifi::WifiMode;
use esp_wifi::{initialize, EspWifiInitFor};
use hal::{clock::ClockControl, peripherals::Peripherals, prelude::*, Delay, IO};
use hal::{systimer::SystemTimer, Rng};

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();
    let clocks = ClockControl::max(system.clock_control).freeze();
    let mut delay = Delay::new(&clocks);

    // setup logger
    // To change the log_level change the env section in .cargo/config.toml
    // or remove it and set ESP_LOGLEVEL manually before running cargo run
    // this requires a clean rebuild because of https://github.com/rust-lang/cargo/issues/10358
    esp_println::logger::init_logger_from_env();
    log::info!("Logger is setup");

    println!("Hello world!");

    let timer = SystemTimer::new(peripherals.SYSTIMER).alarm0;
    let init = initialize(
        EspWifiInitFor::Wifi,
        timer,
        Rng::new(peripherals.RNG),
        system.radio_clock_control,
        &clocks,
    )
    .unwrap();

    let (wifi, _) = peripherals.RADIO.split();
    let (_wifi_interface, mut controller) =
        esp_wifi::wifi::new_with_mode(&init, wifi, WifiMode::Sta).unwrap();
    println!("Device capabilities: {:?}", controller.get_capabilities());

    const SSID: &str = env!("SSID");
    const PASSWORD: &str = env!("PASSWORD");
    let client_config = Configuration::Client(ClientConfiguration {
        ssid: SSID.into(),
        password: PASSWORD.into(),
        ..Default::default()
    });
    println!("Setting configuration: {:?}", &client_config);
    controller.set_configuration(&client_config).unwrap();
    controller.start().unwrap();
    println!("Wifi started: {}", controller.is_started().unwrap());
    controller.connect().unwrap();

    // Set GPIO7 as an output, and set its state high initially.
    let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let button = io.pins.gpio9.into_pull_up_input();
    let mut led = io.pins.gpio7.into_push_pull_output();
    led.set_high().unwrap();

    let mut i = 0;
    loop {
        if button.is_high().unwrap() {
            println!("Loop... {} {:?}", i, controller.is_connected());
            led.toggle().unwrap();
        }
        i += 1;
        delay.delay_ms(500u32);
    }
}
```

I'm using ESP32-C3, and getting bombard with these logs:
```
Loop... 251 Ok(true)
Loop... 252 Ok(true)
Loop... 253 Ok(true)
Loop... 254 Ok(true)
ERROR - RX QUEUE FULL
Loop... 255 Ok(true)
ERROR - RX QUEUE FULL
ERROR - RX QUEUE FULL
Loop... 256 Ok(true)
ERROR - RX QUEUE FULL
Loop... 257 Ok(true)
ERROR - RX QUEUE FULL
ERROR - RX QUEUE FULL
ERROR - RX QUEUE FULL
Loop... 258 Ok(true)
ERROR - RX QUEUE FULL
ERROR - RX QUEUE FULL
Loop... 259 Ok(true)
ERROR - RX QUEUE FULL
ERROR - RX QUEUE FULL
```

Is this a legitimate behavior? If yes, how can I silent these logs? It no, what can be done?

Thanks

bjoernQ, 2023-10-17:
That should probably get logged as WARN not ERROR. Besides that, you should be able to  get rid of all the logs by configurating the logger accordingly or not configuring the logger at all.

It's also possible to configure the queue size and other options: https://github.com/esp-rs/esp-wifi/blob/main/docs/tuning.md

bugadani, 2023-10-17:
I think it's fine for this to be an error. In my experience things rarely recover if the firmware hits this. I'd advise agains just hiding the pain; tuning the queue sizes may be a good way to reduce them.

galah92, 2023-10-17:
I see. The default `rx_queue_size` is 5, what's the best way to tweak it?

bugadani, 2023-10-17:
I set it to 15 for my project. I'm still not satisfied with how my wifi works but at least it's reasonably stable :)

M4tsuri, 2023-10-20:
I'm getting a similar error `Unable to allocate 2308 bytes` on ESP32S2 with the default config when receiving packets at a rate of 1400/s. Managed to get rid of it by changing the config like this:

[image]

The peak heap usage is about 40000 bytes with this configuration.

MabezDev, 2023-10-20:
This probably shouldn't even be a warning, probably a debug statement. Packet memory is now entirely stored in the heap, the warning is about the number of packet headers in the queue, setting this to a low-ish number is fine, because if there is no space in the queue then the wifi stack will just keep retrying. Depending on what you're application needs, you can tune the stack to either be more performant and use more memory (you will need to either increase the heap size or increase the static buffer count) or slower but less memory intensive. See tuning.md for more details.

MabezDev, 2023-10-20:
> I think it's fine for this to be an error. In my experience things rarely recover if the firmware hits this more than once in a while. I'd advise agains just hiding the pain; tuning the queue sizes may be a good way to reduce them.

This was probably true before, but not anymore, I've had this running in the last 24hrs and it will log this error constantly but still work just fine.

galah92, 2023-10-20:
@MabezDev I'm a bit confused.

First saw `ERROR - RX QUEUE FULL` logs. I then set `rx_queue_size = 15`, which removed these logs but made things worse: I got `WARN - Unable to allocate 2312 bytes` logs and WiFi disconnections after ~15 seconds.
I now added `heap_size = 131072` (doubled the heap size) to get again `ERROR - RX QUEUE FULL` logs, but this time without disconnections.

Considering the code I shared above, which basically does nothing in terms of network, I assume I should keep the default config values, and just find a way to silent the `ERROR - RX QUEUE FULL` logs until this log severity will change. Is this correct?

galah92, 2023-10-20:
> ... and just find a way to silent the `ERROR - RX QUEUE FULL` logs ...

I can't seem to find a way to disable specifically the logs coming from the `esp-wifi` crate. Can this be done at all?

MabezDev, 2023-10-20:
I explained it here https://github.com/esp-rs/esp-wifi/issues/291#issuecomment-1772448847, `ERROR - RX QUEUE FULL` is a red herring and should not be an error log.

bjoernQ, 2023-10-23:
You cannot disable logging for specific crates/paths but you can do it the other way around and only enable logging for the things you want to see logs from - see https://github.com/esp-rs/esp-println#logging ( ESP_LOGTARGETS )

MabezDev, 2024-02-20:
The log has since been removed.

closed 2024-02-20 by MabezDev