esp-rs/esp-wifi-sys #305: Too long DHCP discovery

ntqbit, 2023-10-24:
Hello!

I use the latest esp-wifi revision (https://github.com/esp-rs/esp-wifi/commit/55782663e71e1891da106d040cdfadf1bd466c13) att the moment. I run the examples [dhcp.rs](https://github.com/esp-rs/esp-wifi/blob/main/examples-esp32/examples/dhcp.rs) and [embassy_dhcp.rs](https://github.com/esp-rs/esp-wifi/blob/main/examples-esp32/examples/embassy_dhcp.rs) for ESP32 (ESP32-WROOM-32 DevkitC board).

Most of the times DHCP discovery takes more than 9 seconds, sometimes even up to 30 seconds. Sometimes DHCP discovery takes no time and the MCU immediately gets static IP.
But that happens rarely, about 1 in 10 times. In other times I have to wait ~10 seconds.

Although, the wifi connection and transmission/reception speed is ok (at least for MQTT), no significant delays.
Wifi connection takes almost no time, at most a few seconds.

I used two access points to connect: my home wifi and an access point created by my laptop.
With both of them I observe the same problem. The laptop and therefore the access point are located less than a meter away from the ESP32, so I think the signal quality should be great.

I measured the time it takes for the MCU to send the DHCP discovery packet with wireshark:
[image]
(`192.168.12.1` is the DHCP server, the IP of the laptop's interface, `192.168.12.84` is the leased IP for the 

I see that both the discovery and request packets are sent twice. Is that expected behavior?

I need to clarify that the first DHCP discovery packet is usually delayed for ~10 seconds AFTER I see the `Wait to get an ip address` debug line ([source](https://github.com/esp-rs/esp-wifi/blob/55782663e71e1891da106d040cdfadf1bd466c13/examples-esp32/examples/dhcp.rs#L100)), for both embassy async and blocking versions.

The commands I use to run the examples:
```
SSID="ssid" PASSWORD="passwd" cargo run --release --example dhcp --features "wifi"
SSID="ssid" PASSWORD="passwd" cargo run --release --example embassy_dhcp --features "wifi async embassy-net embedded-svc"
```

Do you know what could be the problem?

Thanks!

bugadani, 2023-10-24:
https://github.com/embassy-rs/embassy/pull/2104

ntqbit, 2023-10-25:
> [embassy-rs/embassy#2104](https://github.com/embassy-rs/embassy/pull/2104)

Thank you. That helped.
Indeed, following the pull request, resetting the DHCP socket after STA is connected solves the problem and DHCP discovery happens almost immediately.
However, I found that I also need to add a little delay of about 300 ms before resetting the socket. Otherwise, the problem stays.

So the code working for me (blocking version):
```
....
    println!("{:?}", controller.is_connected());

    let wait_end = current_millis() + 200;
    while current_millis() < wait_end {}
    wifi_stack.reset();

    // wait for getting an ip address
    println!("Wait to get an ip address");
....
```

closed 2023-10-25 by ntqbit