esp-rs/esp-wifi-sys #311: Non-open softAP not supported

bugadani, 2023-10-26:
I know the readme already has a point about not supporting non-open softAP but there is no open issue about the general thing. https://github.com/esp-rs/esp-wifi/issues/277 exists but for WPA2-Enterprise, which is a subset that requires some authmode-specific code.

bugadani, 2023-10-26:
I've tried the obvious: copying config values from esp-idf-svc. Unfortunately, it looks like this issue isn't as easy as a misconfigured softAP.

AnthonyGrondin, 2023-10-27:
My $0.02
Last time I tried to use the SoftAP with an auth method, I fell on the `gettimeofday()` that wasn't implemented. This has been fixed in #304, but this makes me think since a certain number of `extern "c" fn` are being called to initialize it, we should see if there isn't an issue there. It would have normally panicked, if we fell on an unimplemented one, but maybe an implemeted one doesn't return the right value that the precompiled libs expect. I also don't know what's the actual issue after copying the value, so my comment above might be off tracks. 

Can we use this issue to track the implementation, and debugging? And post what bits of code you changed? Or maybe open a Draft PR.

bjoernQ, 2023-10-27:
I guess the problem is that we compile `wpa_supplicant` with TinyCrypt - not mbed-tls: https://github.com/esp-rs/esp-wireless-drivers-3rdparty/blob/master/patch/esp32c3/sdkconfig.defaults

The reason for that is mbed-tls is quite huge and pulls in a lot of stuff from esp-idf

This is also causing #109 

My ultimate dream solution would be #158 ... but that's a huge thing

bugadani, 2023-10-27:
> TinyCrypt 

is that basically a no-op thing that makes things compile but break as soon as anything needs crypto? this is essentially what we seem to be experiencing

bjoernQ, 2023-10-27:
TinyCrypt only includes support for a subset of crypto that is needed for more advanced things: https://github.com/intel/tinycrypt/tree/master/lib/source

Probably the missing parts are no-ops then

Ah seems it's actually not TinyCrypt but some crypto impls in wpa_supplicant itself: https://github.com/espressif/esp-idf/blob/b4268c874a4cf8fcf7c0c4153cffb76ad2ddda4e/components/wpa_supplicant/CMakeLists.txt#L104-L161

bjoernQ, 2023-10-27:
But the thing about the crypto is just guessing currently - probably we need to test that with an ESP-IDF example and see what config options are really needed

noahbliss, 2023-11-21:
Just bumping this up in the fresh list. :P

noahbliss, 2024-02-19:
Is there a workaround for this? Even one that imports an inconveniently-large lib?

AnthonyGrondin, 2024-02-20:
I think the best bet would be to get https://github.com/esp-rs/esp-wifi/pull/394 working, and implement the crypto functions.