esp-rs/esp-wifi-sys #315: Losing connection while downloading large file may result in unrecoverable state

bugadani, 2023-10-27:
The "while downloading large file" may not be necessary but it's how I'm able to reproduce this issue the most reliably. It looks like the TX queue fills up, and never gets cleared afterwards.

bjoernQ, 2023-10-27:
Might be a new thing?

How large is the file? During my tests to improve the throughput I think I used files >2M. Also in the very beginning I had an MQTT temperature logger running for more than one week without error recovery. (But with the non-async API... There was no async back then)

bugadani, 2023-10-27:
I'm downloading and discarding a firmware update, around 1MB, as part of a throughput test. The key is losing connection for more than a couple seconds. The indication that this happens is that the download speed drops to 0 and stays 0. 1-2 seconds of this is surviveable, but in this state as soon as the TX queue fills up with frames, the driver can't seem to recover.

bugadani, 2023-10-31:
My app has been running for 3.5 hours in a loop without the issue, so I assume #318 has fixed it.

closed 2023-10-31 by bugadani